hku-mars/FAST-LIVO #1: Code release!

aditdoshi333, 2022-03-08:
Hello,

I am quite excited to try out your work and it is based on FASTLIO 2 and visual odometry. Is there any approximate code release date?

xuankuzcr, 2022-04-19:
Hi, aditdoshi333
Sorry for the late reply. We will release our code, dataset and hardware system in early May due to the limited free time recently. Thanks for your interest and attention to our work!

FPSychotic, 2022-05-02:
Thanks very much, I'm specially interested in the synchronization and the gps part.
I hope that part get release too. 
Because I'm trying to learning how that things works, I don't understand very well, but as it is done with stm32, maybe I could understand it and make my own for realsenses with gps and external IMU etc. That looks very interesting for me

closed 2022-11-14 by XW-HKU