hku-mars/FAST-LIVO #4: Comparing with R3Live

Camilochiang, 2022-05-01:
When comparing the resolution of FAST-LIVO to R3live from your videos, is it clear for me that FAST-LIVO is doing a better work . Have been this difference been quantified?
**R3Live**
[image]
**FAST-LIVO***
[image]

xuankuzcr, 2022-05-11:
Hi, Camilo
Thanks for your attention! First of all, we can't determine whose effect is better only from the performance of the two systems in this scenario. This is because the two bags come from different data acquisition devices, different lighting conditions, different acquisition routes and different camera parameters, which will have a huge impact on the final reconstruction results. Secondly,  we haven't compared the differences between them quantitatively, maybe we'll do this experiment later. However, maintaining high accuracy and robustness on the premise of high speed is what we claim in our system, which doesn't contradict R3LIVE. 
Actually, for this non-degenerate structured scenario, their accuracy should be similar, benefited from the high-precision odom provided with LIO subsystem (FAST-LIO2).

Camilochiang, 2022-05-11:
Thanks for your answer!

xuankuzcr, 2022-12-15:
[Here](https://www.bilibili.com/video/BV1oA41197fe/?spm_id_from=333.337.search-card.all.click) is someone who compared the colored point clouds with the same data, the same computer configuration and the same perspective. You can take a look and compare them qualitatively.

FPSychotic, 2022-12-15:
I find here two observations, R3Live can rise the definition of the vertex colour, I can find on my hand held scanner both can get the same coloured quality.
I find R3live is more CPU power hungry in same quality range, but Fast LIVO will  end with 16Gb RAM in 3 min, R3 Live can keep working for at least 15mim.
For other side there is not a real or practical utility that the point cloud could or not be more defined in n colour , as the map in realtime cannot be saved.
It looks the R3live VIO subsystem is less dependent of the frame rate, camera quality or global shutter or not, fastlivo does, so with my d435 looks R3Live gives better tracking in indoors and low textured areas, didn't test in outdoors. 
This is what I find, but other hardware, computer, setups could give different result in a way or other

closed 2023-01-12 by Camilochiang

FPSychotic, 2023-01-12:
Just let know and longer test fast-livo has effectively better definition in the coloured pointcloud. And it is more solid VO tracking. Is quite nice, but cannot save the map directly and the ram it eats make it difficult to work in realtime. But I think the Dev is brilliant and the software does what he wanted, is better that I said in my first comment based in indoor tests