hku-mars/FAST-LIVO #6: Difference FAST LIVO and R3LIVE

antun008, 2022-06-19:
Hello,
What is difference between FAST LIVO and R3LIVE?

Best Regards,
Antun

xuankuzcr, 2022-06-20:
Hi antun088,
The VIO subsystem in R3LIVE calculates the photometric error between the RGB-colored 3D points in the map and the current frame, while FAST-LIVO uses the multi-layer pyramid patches in the map. R3LIVE is more dense than our sparse direct method. Naturally, our framework will run faster on the basis of keeping accuracy. Of course, there are many differences in detail and framework design due to methodological differences, such as strategies for updating the visual map and so on.

mmhhkk123, 2022-06-21:
@xuankuzcr Hi, thanks for your work and contribution! By the way, will the source codes be released recently?

xuankuzcr, 2022-06-21:
@mmhhkk123 Thanks for your patience. I'm really sorry for the delay. I noticed that the result of IROS2022 will be published on June 30. If it is accepted, I will release the source code.

closed 2022-06-28 by antun008