hku-mars/FAST-LIVO #9: Whether the improvement on NTU-VIRAL datasets is still significant after LiDAR and IMU timeoffset correction

BITcats, 2022-09-14:
Thank you for your excellent work! 

The ouster pointcloud and IMU messages in NTU-VIRAL  datasets exist synchronization issues.  Please refer to [this](https://github.com/ntu-aris/ntu_viral_dataset)

After regularizing the ouster pointcloud and imu topics in NTU datasets, I retest FAST-LIO2.0 on it and the average absolute pose error is about 0.03m.  When I test a LIVO system based on FastLIO2.0 implemented by myself, I find the VIO subsystem can not improve performance significantly. The average absolute pose error of the whole LVIO system on NTU-VIRAL datasets is about 0.029m.

It would be very helpful if you could provide me with the experiment results on corrected NTU-VIRAL datasets.

Camilochiang, 2022-09-27:
Hei @BITcats sorry to bother you. Is your LIVO system available? So far all the algorithms from hku have fail in an agricultural env and I would like to test FAST-LIVO but is still not available and not sure if this research group will give it priority...

lyeemax, 2022-10-26:
@BITcats ,as mentioned in ntu-aris:
> 12/07/2022: The ouster pointcloud and IMU messages are found to jitter due to synchronization issues. A script to regularize the ouster pointcloud and imu topics can be downloaded [here](https://github.com/ntu-aris/ntu_viral_dataset>/blob/gh-pages/utils/restamp.py). 

I implement my LIVO using left camera and horizontal lidar and imu on NTUVIRAL dataset. In my opinion, VIO subsystem would not improve LVIO system significantly, the possible reasons are: 
1.the image frequency of NTU_VIRAL dataset is 10Hz, since lidar is 10Hz, the advantage of high camera rate  has not been utilized.
2.the observabilty of gravity and bias might be the potential factor that affects the LIO system greatly, however, visual features are scale-free.

closed 2023-01-14 by xuankuzcr