hku-mars/FAST-LIVO #12: 有关测试自己数据集的问题

hr2894235132, 2022-11-14:
您好！看到fast-livo代码开源非常兴奋，非常感谢您将代码开源供我们参考学习！！！
但是，我在使用自己的手持设备（**mid70+realsence D455相机及相机内置imu（事先已经进行了内外参的标定）**）测试的时候，遇到了和测试R3live一样的问题，静止时一切安好，只要一动起来里程计就漂了，无法正常建图。（但是在之前测试fast-lio2时，效果很好。）
# 室内测试：
[image]
[image]
[image]
## 室内config：
```
feature_extract_enable : 0
point_filter_num : 1
max_iteration : 10
dense_map_enable : 1
filter_size_surf : 0.05 # 建议室内：0.05~0.15；室外：0.3~0.5
filter_size_map : 0.15 # 建议室内：0.15~0.3；室外：0.4~0.5
cube_side_length : 20
debug : 0
grid_size : 40
patch_size : 8
img_enable : 1
lidar_enable : 1
outlier_threshold : 300 # 78 100 156 建议较暗的场景为50~250，较亮的场景为500~1000。该值越小，vio子模块越快，但抗退化能力越弱。
ncc_en: false # ??
ncc_thre: 0 # ??
img_point_cov : 100 # 1000 The covariance of photometric errors per pixel.
laser_point_cov : 0.001 # 0.001 The covariance of point-to-plane redisual per point.
cam_fx: 426.551167166986
cam_fy: 426.6142447926181
cam_cx: 429.1314081919893
cam_cy: 247.64043896840604

common:
    lid_topic:  "/livox/lidar"
    imu_topic:  "/camera/imu"

preprocess:
    lidar_type: 1 # Livox Avia LiDAR
    scan_line: 1
    blind: 0.05 # blind x m disable

mapping:
    acc_cov_scale: 100
    gyr_cov_scale: 10000
    fov_degree:    70
    # extrinsic_T: [ 0.04165, 0.02326, -0.0284 ]
    # extrinsic_R: [ 1, 0, 0,
    #                0, 1, 0,
    #                0, 0, 1]
    extrinsic_T: [ -0.017020, 0.085815, -0.024827 ]
    extrinsic_R: [ 0.046790, -0.998811, 0.013708,
                   -0.023212, -0.014807, -0.999621,
                   0.998635, 0.046454, -0.023877]

camera:
    # img_topic: /usb_cam/image_raw
    # img_topic:  /camera/image_color
    img_topic: "/camera/color/image_raw"
    #xiyuan
    # lidar to camera
    # Rcl: [0.0110805,-0.999823,0.0151929,
    #      -0.0410198,-0.0156355,-0.999036,
    #       0.999097,0.0104466,-0.0411858]
    # Pcl: [0.0183, 0.0762623, -0.0305996]
    # camera to lidar
    Rcl: [0.0110805, -0.0410198, 0.999097,
         -0.999823, -0.0156355, 0.0104466,
          0.0151929, -0.999036, -0.0411858]
    Pcl: [0.0334975, 0.0198088, 0.0746505]
```
# 室外测试：
[image]
## 室外config：
只修改了两个降采样的参数
```
filter_size_surf : 0.3 
filter_size_map : 0.5 
```

**请问一下您知道什么原因吗？或者有没有一些建议呢？**

hr2894235132, 2022-11-15:
> 首先需要确定相机和imu之间是否同步好，realsense的imu的时间戳似乎有问题。

您好，感谢您的回复！
我们使用realsence内置的imu，和相机之间的同步的话应该是同步好的，imu和Mid-70的同步也已经做好了，对lidar数据已经进行了时间同步。

XW-HKU, 2022-11-15:
> > 首先需要确定相机和imu之间是否同步好，realsense的imu的时间戳似乎有问题。
> 
> 您好，感谢您的回复！ 我们使用realsence内置的imu，和相机之间的同步的话应该是同步好的，imu和Mid-70的同步也已经做好了，对lidar数据已经进行了时间同步。

我们测过，realsense的内置imu和realsense的pointcloud时间戳有的时候并没有对齐上。

hr2894235132, 2022-11-16:
> > > 首先需要确定相机和imu之间是否同步好，realsense的imu的时间戳似乎有问题。
> > 
> > 
> > 您好，感谢您的回复！ 我们使用realsence内置的imu，和相机之间的同步的话应该是同步好的，imu和Mid-70的同步也已经做好了，对lidar数据已经进行了时间同步。
> 
> 我们测过，realsense的内置imu和realsense的pointcloud时间戳有的时候并没有对齐上。

好的，那我再去检查确认一下，感谢您的耐心回复！

Fellowflyingsky, 2022-11-16:
> > > 首先需要确定相对机和imu之间是否同步好，realsense的imu的时间触摸好像有问题。
> > 
> > 
> > Realsence内置的内置内置好
> 
> 我们测试过，realsense 的内部设置 imu 和 realsense 的点云时间有的时候并没有对齐上。

您好：这套代码必须要做硬件同步么，用之前r3跑的自己的数据集可以直接 拿过来用么

XW-HKU, 2022-11-16:
> > > > 首先需要确定相对机和imu之间是否同步好，realsense的imu的时间触摸好像有问题。
> > > 
> > > 
> > > Realsence内置的内置内置好
> > 
> > 
> > 我们测试过，realsense 的内部设置 imu 和 realsense 的点云时间有的时候并没有对齐上。
> 
> 您好：这套代码必须要做硬件同步么，用之前r3跑的自己的数据集可以直接 拿过来用么

必须硬件同步。

hr2894235132, 2022-11-16:
您好，您用自己数据集成功测试了R3live吗？

linzs-online, 2022-11-21:
> > 首先需要确定相机和imu之间是否同步好，realsense的imu的时间戳似乎有问题。
> 
> 您好，感谢您的回复！ 我们使用realsence内置的imu，和相机之间的同步的话应该是同步好的，imu和Mid-70的同步也已经做好了，对lidar数据已经进行了时间同步。

您好，我想请问一下您是怎么把IMU和Mid-70做同步的？似乎realsense相机没有外部时钟触发接口。我们也遇到了同样的问题，想请教一下您是怎么解决的，感谢~

lg395381306, 2022-11-24:
请问雷达和FLIR相机怎么做硬件同步呢？

closed 2023-01-14 by xuankuzcr

NianHu, 2024-02-01:
> 您好！看到fast-livo代码开源非常兴奋，非常感谢您将代码开源供我们参考学习！！！ 但是，我在使用自己的手持设备（**mid70+realsence D455相机及相机内置imu（事先已经进行了内外参的标定）**）测试的时候，遇到了和测试R3live一样的问题，静止时一切安好，只要一动起来里程计就漂了，无法正常建图。（但是在之前测试fast-lio2时，效果很好。）
>
> # 室内测试： [...]

您好，我最近用D455和Mid360遇到了同样的问题，请问您后来解决了吗

NianHu, 2024-02-01:
您好，我最近用D455和Mid360遇到了同样的问题，请问您后来解决了吗

hopenessQAQ, 2024-06-24:
> > > > > 首先需要确定相对机和imu之间是否同步好，realsense的imu的时间触摸好像有问题。
> > > > 
> > > > 
> > > > Realsence内置的内置内置好
> > > 
> > > 
> > > 我们测试过，realsense 的内部设置 imu 和 realsense 的点云时间有的时候并没有对齐上。
> > 
> > 
> > 您好：这套代码必须要做硬件同步么，用之前r3跑的自己的数据集可以直接 拿过来用么
> 
> 必须硬件同步。

您好，如果通过软件同步方式可以运行FAST-LIVO么

hr2894235132, 2024-06-24:
> > > > > > 首先需要确定相对机和imu之间是否同步好，realsense的imu的时间触摸好像有问题。
> > > > > 
> > > > > 
> > > > > Realsence内置的内置内置好
> > > > 
> > > > 
> > > > 我们测试过，realsense 的内部设置 imu 和 realsense 的点云时间有的时候并没有对齐上。
> > > 
> > > 
> > > 您好：这套代码必须要做硬件同步么，用之前r3跑的自己的数据集可以直接 拿过来用么
> > 
> > 
> > 必须硬件同步。
> 
> 您好，如果通过软件同步方式可以运行FAST-LIVO么

可以的

hopenessQAQ, 2024-06-24:
> > > > > > > 首先需要确定相对机和imu之间是否同步好，realsense的imu的时间触摸好像有问题。
> > > > > > 
> > > > > > 
> > > > > > Realsence内置的内置内置好
> > > > > 
> > > > > 
> > > > > 我们测试过，realsense 的内部设置 imu 和 realsense 的点云时间有的时候并没有对齐上。
> > > > 
> > > > 
> > > > 您好：这套代码必须要做硬件同步么，用之前r3跑的自己的数据集可以直接 拿过来用么
> > > 
> > > 
> > > 必须硬件同步。
> > 
> > 
> > 您好，如果通过软件同步方式可以运行FAST-LIVO么
> 
> 可以的

感谢 我用ROS提供的方法尝试一下

FPSychotic, 2024-06-24:
Zw