hku-mars/FAST-LIVO #14: Livox driver and RVIZ visualization

FPSychotic, 2022-11-14:
Hi, which is the Livox driver is needed to use, the modified HKU as R3Live or the stock one and what launch I need to use, lidar.launch or lidar_msg.launch. 

I'm using real hardware, the same that works with fast_lio and R3live without issue,  but my issue with fast_livo is in RVIZ I cannot see the transforms and frame published, there is not lidar frames at all. There is not published TF as world, map, etc.
I think the software is working , I post picture of the console, no error messages and something is doing.

[image]
[image]

FPSychotic, 2022-11-15:
this is the TF tree when run FastLio , as you can see I have TF from it
[image]

MatCapLynx, 2022-11-15:
To enable tf, you should uncomment lines 920 to 929 in `/src/laserMapping.cpp` ([here](https://github.com/hku-mars/FAST-LIVO/blob/5f9e62209ca175d00bdac668965fde8eca8eda1b/src/laserMapping.cpp#L920)).  
You don't have any points in `/cloud_registered_rgb` but there should be some in `/cloud_registered`. This is probably due to a bad extrinsic transform between camera and lidar (`camera/Rcl` and `camera/Pcl`).

MatCapLynx, 2022-11-15:
No it's not the same, I don't know why. You should use the output from https://github.com/hku-mars/livox_camera_calib.  
The matrix should look like : 
```
-0.0568933,-0.998377,-0.00262438,-0.0230558
0.00675769,0.00224349,-0.999975,0.0176824
0.998357,-0.0569096,0.00661908,-0.0893782
0,0,0,1
```

FPSychotic, 2022-11-15:
ohh, thanks so much, I can see already the the /cloud registered in the FastLIVO Rviz config after uncoment the lines and get transforms.
I will try now to change the extrinsic matrix as you said, but even with that I afraid I won't get a map as there is no TFs /world and /body, there is some lines commented in the same file, any tip about that? Thanks, I know where to look, but I don't know what change as I don't know programming.
I'm going to change to image_raw (realsense uncompresed) as the republished in the launch  (compressed ) has a lot of latency. I hope that won't give any issue.
[image]

FPSychotic, 2022-11-15:
[image]
[image]

I written my matrix as above but maybe I'm doing wrong, as still nothing in registred rgb prointcloud.
I using the default driver with the msg.launch, lidar is a AVIA and everything is sychronized via ntpt, this setup works in r3live.

MatCapLynx, 2022-11-15:
These extrinsics should be filled in Rcl and Pcl (camera to lidar) not extrinsic_R and extrinsic_T (lidar to IMU).

The 4x4 extrinsic matrix corresponds to:
```
R00 R01 R02 T0
R10 R11 R22 T1
R20 R21 R22 T2
0   0   0   1
```

FPSychotic, 2022-11-15:
> These extrinsics should be filled in Rcl and Pcl (camera to lidar) not extrinsic_R and extrinsic_T (lidar to IMU).
> 
> The 4x4 extrinsic matrix corresponds to:
> 
> ```
> R00 R01 R02 T0
> R10 R11 R22 T1
> R20 R21 R22 T2
> 0   0   0   1
> ```
Sorry so much by disturb you, but I placed the matrix in the right place, I did as you said (I think as programming, matrix etc is a little difficult for me, but I will look and search to understand it)
I did with the R3LIVE  too , which fit in numbers distribution with the Fast-livo config file, as yours have 4 numbers and Fast-livo and R3LIVE have 3, it is like your matrix is 4x4 but their is 4x3.
Anyways, after try both configurations, it still doesn't work. I think World, body TF are no active, in Rviz , the registered_rgb don't allow even select RGB8 visualization so I think the RGB visualization is even disabled from code or from the Rviz config file.
[image]
[image]
[image]

FPSychotic, 2022-11-16:
> You didn't put the good configuration, you should put:
> 
> ```
> Rcl: [0.0302, -0.999, -0.010,
>         0.007,   0.011,  0.031
>         0.999,   0.030,  0.007]
> Pcl: [0.001, 0.031, -0.037]
> ```

Ohh, sorry by lack of understanding on this, thank you so much but write right thing. I will try in a few hours

FPSychotic, 2022-11-16:
Hi, after test the matrix and having odometry (not sure if from LIO, VIO or both) 
and being able to see the registered cloud, see the camera image_raw and can map with it , registered_cloud_rgb don't show anything. 
What I can see the camera's TFs are not connected with the generated by Fast-livo.
[image]

FPSychotic, 2022-11-21:
Hi @xuankuzcr, Im sorry by quote you,with the info above could you give me any hint to make this work the mapping and point cloud colourisation?. Now is working registered and effected, odometry aft mapped to init is working too and get good odometry, only the mapping part is not working, Im using the handheld variation instead a drone. My problem I think comes from transforms as camera and odometry are in different tree as shown above and I can see in the lasermappIing.cpp some transform lines un-commented, should I comment some lines? I tried both livox drivers, the r3live modified one and the normal one and both look work, if you can advice me which should |I use would be great.

Thanks