hku-mars/FAST-LIVO #17: A question about ikfom

gisbi-kim, 2022-11-16:
Thanks for the great code, and a kindly sharing. 
I have a short question. I read the fastlivo code as well as r2live, and I noticed ikfom does not being used no longer, but replaced with a simple lines of Eigen computations. Any reasons (and pros/cons)?

xuankuzcr, 2022-11-17:
Ikfom is an accurate implementation of EKF on manifold. Handwritten ekf will make some approximations, such as the derivation of the rotation matrix and the projection between update and prediction. In fact, ikfom is used to reduce the difficulty of manually implementing a specific system from scratch, which can be deployed more easily. If we have manually implement an available ekf, ikfom is not necessary.

gisbi-kim, 2022-11-17:
So clear. Thanks!!

closed 2022-11-17 by gisbi-kim

ToutDonner, 2023-03-22:
您好，我在测试自己的数据集时发现FAST-LIO要比FAST-LIVO(关闭VIO子系统)精度高，请问EKF实现方法不同会导致精度的差异吗？

xuankuzcr, 2023-03-29:
这个应该和IKFOM或者其他的实现没关系，单纯因为LIO的版本问题，公开的FAST-LIO可能解决了一些bug已经和LIVO里的不是一个版本了。

ToutDonner, 2023-03-30:
> 这个应该和IKFOM或者其他的实现没关系，单纯因为LIO的版本问题，公开的FAST-LIO可能解决了一些bug已经和LIVO里的不是一个版本了。

了解，谢谢！