hku-mars/FAST-LIVO #21: Hardware time

Tomato1107, 2022-11-18:


Tomato1107, 2022-11-20:
[image]

closed 2022-11-28 by Tomato1107

Camilochiang, 2022-12-01:
@Tomato1107, If im not wrong  what is happening, if you have a correct hardware sync,  is that you are synchronizing the start of the measurement and not when you are reciving the ROS message.  That means that the camera take significantly longer to capture the image compared with the lidar and IMU. Of course this depend of the camera that you are using, ROS package and the computer speed. Why do you want to have the messages arriving at the same time?

Camilochiang, 2022-12-01:
You don’t need to run any algorithm to check time stamps . 
```
roscore #in one tab of terminal
rqt_graph YOURBAG.bag 
```
please run it with your bags and you will get always the same time stamps . Is your image bellow always the same bag?

Tomato1107, 2022-12-02:
I recorded several rosbags, the time difference also seemed to be different
when I run fastlivo. It really takes some time for the camera to deal with
the image, currently I just use the 10Hz signal to trigger it.  You can
also compare different rosbags to check the results.

xuankuzcr, 2023-12-06:
The synchronization scheme has been updated, as mentioned in Issue #44
@Tomato1107 @Camilochiang