hku-mars/FAST-LIVO #23: Hardware Time Sync Issues

satyajitghana, 2022-11-23:
Hi,

I tried your bag files, and tried to get the difference between the camera timestamp and the lidar timestamp. It gives exactly 0.000000000.  How is this possible? Did you modify the bag files and replace camera timestamp with lidar timestamp? Even in nanoseconds there is no difference.

jay80253504, 2022-11-25:
same confusion

kahowang, 2022-12-01:
same confusion

xuankuzcr, 2022-12-12:
If you can ensure that the frame header of the camera and the LiDAR at the phisical trigger time is corresponding, you can replace the camera timestamp with the LiDAR timestamp, which eliminates the impact of network cable or usb transmission delay and ROS delay.

closed 2023-01-14 by xuankuzcr

xuankuzcr, 2023-12-06:
@satyajitghana @jay80253504 
The synchronization scheme has been updated, as mentioned in Issue #54