hku-mars/FAST-LIVO #25: "Add 0 3D points" when testing with R3LIVE data.

OliverShaoPT, 2022-11-29:
First of all, thanks for sharing your great work!
I want to test FAST-LIVO with some long bags, such as the R3LIVE datasets.
[image]
After adjust the parameter in yaml and launch file, "Add 0 3D points " occured.

[ INFO ]: get img at time: 1630286393.589510.
[ INFO ]: get point cloud at time: 1630286393.584056.
[ LIO ]: Raw feature num: 5325 downsamp num 4684 Map num: 11234.
[ LIO ]: Using multi-processor, used core number: 4.
[ LIO ]: time: fov_check: 0.000000 fov_check and readd: 0.000911 match: 0.002030 solve: 0.000292  ICP: 0.013996  map incre: 0.008080 total: 0.012388 icp: 0.002380 construct H: 0.000186.
[ VIO ]: Raw feature num: 5325.
**[ VIO ]: Add 0 3D points.**
[ VIO ]: time: addFromSparseMap: 0.000002 addSparseMap: 0.000274 ComputeJ: 0.000000 addObservation: 0.000001 total time: 0.000277 ave_total: 0.000277.

It seems to track raw feature successfully, but no points added. Here is my yaml and launch file.

[image]
[image]

Waitting for your reply.

xuankuzcr, 2022-12-12:
The dataset in R3LIVE is not hard synchronized, so FAST-LIVO cannot run directly. You need to calculate the time offset between the camera and IMU in advance and compensate it into the system.

OliverShaoPT, 2022-12-20:
> The dataset in R3LIVE is not hard synchronized, so FAST-LIVO cannot run directly. You need to calculate the time offset between the camera and IMU in advance and compensate it into the system.

Thanks for your reply.  After compensate the time offset, the Visual odometry and map can work. Beside the time offset issue, the camera extrinsic value of R3LIVE and FAST-LIVO can be confusing, one is camera_ext_IMU, the other is camera_to_IMU.

closed 2022-12-20 by OliverShaoPT

xuankuzcr, 2022-12-20:
Yes, their external parameters are contrary to each other.

Camilochiang, 2022-12-21:
@OliverShaoPT Could you please comment how did you include the time offset? Im curious. Thanks!

PengKunPROO, 2024-01-13:
Hi, My dataset is not hard sync too,could you tell me your ways to compensate the time offset between IMU and camera? Does it need to calib or other ways to solve it?

hr2894235132, 2024-02-01:
> Thanks for your reply. After compensate the time offset, the Visual odometry and map can work. Beside the time offset issue, the camera extrinsic value of R3LIVE and FAST-LIVO can be confusing, one [...]

Hello, can you share how you used fast-livo to run the degenerate sequence in the R3live dataset?