hku-mars/FAST-LIVO #28: how to save the map?

FPSychotic, 2022-12-11:
hi , I can't find how save the map, does anyone know? thanks

FPSychotic, 2022-12-12:
Hi, thanks by answer, I did and also tried add a folder address, I created a folder, even copied the Fastlio2 lines in the config file, nothing worked, at least I couldn't find any file. I saw a .CSV and I did open it with cloud compare , it shown only a line, it is not a pointcloud

xuankuzcr, 2022-12-12:
Maybe the version change causes problems with this function. Do you need the RGB-colored point cloud or intensity-colored point cloud? Maybe I can help you develop this function in my spare time.

Camilochiang, 2022-12-12:
Hei @xuankuzcr , saving the RGB and Intensity in one fill will be usefull to be honest! Do you think that is feasable?

FPSychotic, 2022-12-12:
> Maybe the version change causes problems with this function. Do you need the RGB-colored point cloud or intensity-colored point cloud? Maybe I can help you develop this function in my spare time.

Ohh, thank you by offering, would be nice have it, probably many users now and in the future would/be happy with it, but I could not accept you expend you spare time in something that you don't need.  If one day you need  it and do it, I will happy to use it then.

Thank you so much again.

xuankuzcr, 2022-12-12:
> > Maybe the version change causes problems with this function. Do you need the RGB-colored point cloud or intensity-colored point cloud? Maybe I can help you develop this function in my spare time.
> 
> Ohh, thank you by offering, would be nice have it, probably many users now and in the future would/will be happy with it, but I could not accept you expend you spare time in something that you don't need only for me. If one day you need it and do it, I will happy to use it then.
> 
> Thank you so much again.

Thanks for your advice and understanding.

closed 2022-12-12 by FPSychotic