hku-mars/FAST-LIVO #31: 【求助】Double free or corruption (out)报错，终止

Noahcuptea, 2022-12-14:
尊敬的港大团队，您好，

在测试数据集或者收集数据都会出现报错：

[ LIO ]: Raw feature num: 1417 downsamp num 161 Map num: 152.
[ LIO ]: Using multi-processor, used core number: 4.
double free or corruption (out)
[laserMapping-1] process has died  ... 

节点是在接受lidar和imu数据后就会报错，情况如下：
https://user-images.githubusercontent.com/45952143/207554474-f0282eb6-b0f2-4e81-9771-207f330cc036.mp4

lidar和imu的硬同步应该没有问题，因为FAST-LIO可以跑，报错时候的log files已上传，您过目。
[rosout.log](https://github.com/hku-mars/FAST-LIVO/files/10226231/rosout.log)
[rosout-1-stdout.log](https://github.com/hku-mars/FAST-LIVO/files/10226232/rosout-1-stdout.log)
[rviz-3-stdout.log](https://github.com/hku-mars/FAST-LIVO/files/10226233/rviz-3-stdout.log)
[imu-1.log](https://github.com/hku-mars/FAST-LIVO/files/10226234/imu-1.log)
[master.log](https://github.com/hku-mars/FAST-LIVO/files/10226235/master.log)
[roslaunch-cpc-MoreFine-S500-319789.log](https://github.com/hku-mars/FAST-LIVO/files/10226236/roslaunch-cpc-MoreFine-S500-319789.log)
[roslaunch-cpc-MoreFine-S500-320153.log](https://github.com/hku-mars/FAST-LIVO/files/10226237/roslaunch-cpc-MoreFine-S500-320153.log)
[roslaunch-cpc-MoreFine-S500-320331.log](https://github.com/hku-mars/FAST-LIVO/files/10226238/roslaunch-cpc-MoreFine-S500-320331.log)

xuankuzcr, 2022-12-14:
试试gdb debug模式通过这样修改 xxx.launch
<node pkg="fast_livo" type="fastlivo_mapping" name="laserMapping" output="screen" launch-prefix="gdb -ex run --args">.
 然后看看终端的报错位置

Noahcuptea, 2022-12-23:
您好，下图是报错信息，现在往哪个方向整？

[image]

Noahcuptea, 2022-12-26:
> 试试gdb debug模式通过这样修改 xxx.launch `<node pkg="fast_livo" type="fastlivo_mapping" name="laserMapping" output="screen" launch-prefix="gdb -ex run --args">` 然后看看终端的报错位置

您好，您的环境都用的什么？我搜索了港大团队过去的issues，我怀疑是环境问题，您可以把您的环境列出来么？我按照您的环境做一遍系统, 甚至做一份docker。您可以这么列：ubuntu 18.04 - ros_melodic - pcl 1.9 - eignen 3.3.4 - opencv 3.2 - XXX

Noahcuptea, 2022-12-28:
您好，以下是gdb调试输入bt后所获信息的截图和代码（一样的），您过目

[image]
[image]

`[ LIO ]: Raw feature num: 6335 downsamp num 4224 Map num: 2890.
[ LIO ]: Using multi-processor, used core number: 4.
[New Thread 0x7fff77fff700 (LWP 1052713)]
[New Thread 0x7fff777fe700 (LWP 1052714)]
[New Thread 0x7fff76ffd700 (LWP 1052715)]
double free or corruption (out)
--Type <RET> for more, q to quit, c to continue without paging--bt

Thread 1 "fastlivo_mappin" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50 ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff47cd859 in __GI_abort () at abort.c:79
#2  0x00007ffff483826e in __libc_message
    (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4962298 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff48402fc in malloc_printerr
    (str=str@entry=0x7ffff4964670 "double free or corruption (out)")
    at malloc.c:5347
#4  0x00007ffff4841fa0 in _int_free
    (av=0x7ffff4997b80 <main_arena>, p=0x55555c58b710, have_lock=<optimized out>)
    at malloc.c:4314
#5  0x00007ffff4e3841a in Eigen::internal::aligned_free(void*)
    (ptr=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:745
#6  Eigen::aligned_allocator<pcl::PointXYZINormal>::deallocate(pcl::PointXYZINormal*, unsigned long) (this=<optimized out>, p=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:747
#7  std::allocator_traits<Eigen::aligned_allocator<pcl::PointXYZINormal> >::deallocate(Eigen::aligned_allocator<pcl::PointXYZINormal>&, pcl::PointXYZINormal*, unsigned long) (__a=..., __n=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/9/bits/alloc_traits.h:332
#8  std::_Vector_base<pcl::PointXYZINormal, Eigen::aligned_allocator<pcl::PointXYZINormal> >::_M_deallocate(pcl::PointXYZINormal*, unsigned long)
    (this=<optimized out>, __n=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/9/bits/stl_vector.h:351
#9  std::_Vector_base<pcl::PointXYZINormal, Eigen::aligned_allocator<pcl::PointXYZINormal> >::~_Vector_base() (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/stl_vector.h:332
#10 std::vector<pcl::PointXYZINormal, Eigen::aligned_allocator<pcl::PointXYZINormal> >::~vector() (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/stl_vector.h:680
#11 KD_TREE::Add_Points(std::vector<pcl::PointXYZINormal, Eigen::aligned_allocator<pcl::PointXYZINormal> >&, bool)
    (this=0x555555629520 <ikdtree>, PointToAdd=std::vector of length 4224, capacity--Type <RET> for more, q to quit, c to continue without paging--bt
 4224 = {...}, downsample_on=true)
    at /home/cpc/catkin_ws/src/FAST-LIVO/include/ikd-Tree/ikd_Tree.cpp:401
#12 0x000055555558efad in main(int, char**)
    (argc=<optimized out>, argv=<optimized out>)
    at /home/cpc/catkin_ws/src/FAST-LIVO/src/laserMapping.cpp:1746`

Noahcuptea, 2023-01-05:
做好了，库之间一定要配套，我做了如下改动：
- ubuntu 20.04 -> 18.04 （ros neotic -> melodic）
- 重装系统的分区 swag 与实际物理内存一致，系统里显示15左右gib=16gb
- eigen 3.40 -> 3.3.4-4
- 加了vtk 7.1.1
- pcl 1.13.0 -> 1.8.1
- opencv 4 -> opencv 3.4.19
- 还装了显卡配套驱动

推荐软件 timeshift，一个系统备份还原工具，方便restore至合适的节点

closed 2023-01-05 by Noahcuptea

QingKe-UP, 2024-07-19:
@Noahcuptea 您好，请问您后面排查出是哪里出问题了吗？或者您是如何解决这个问题的？

zhao-zhibo, 2025-04-28:
@Noahcuptea @QingKe-UP 我也遇到了这个问题，现在报错是
```
double  free  or  corruption(out)    
 Thread 1 "lio_sam_6axis_m" received signal SIGABRT, Aborted.   
  __GI_raise (sig=sig@entry=6)  at ../sysdeps/unix/sysv/linux/raise.c:50  
 50   ../sysdeps/unix/sysv/linux/raise.c:  没有那个文件或目  
```

然后我还用gdb打印了栈信息，可是报错里面的内容是gtsam中优化函数的问题，我已经把gtsam重新安装了，安装的gtsam版本是4.0.0，Eigen是3.3.7版本。可是还是，没有解决问题。
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff67be859 in __GI_abort () at abort.c:79
#2  0x00007ffff6829266 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff6953298 "%s\n") at ../sysdeps/posix/libc_fatal.c:156
#3  0x00007ffff68312fc in malloc_printerr (str=str@entry=0x7ffff6955670 "double free or corruption (out)") at malloc.c:5347
#4  0x00007ffff6832fa0 in _int_free (av=0x7ffff6988b80 <main_arena>, p=0x55555575a930, have_lock=<optimized out>) at malloc.c:4314
#5  0x00007ffff723285d in gtsam::NoiseModelFactor::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
#6  0x00007ffff7233b3c in gtsam::(anonymous namespace)::_LinearizeOneFactor::operator()(tbb::blocked_range<unsigned long> const&) const () from /usr/local/lib/libgtsam.so.4
#7  0x00007ffff7233f1d in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, gtsam::(anonymous namespace)::_LinearizeOneFactor, tbb::auto_partitioner const>::execute() () from /usr/local/lib/libgtsam.so.4
#8  0x00007ffff6d06545 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#9  0x00007ffff6d0680f in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#10 0x00007ffff6d03b68 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
#11 0x00007ffff7235c9f in gtsam::NonlinearFactorGraph::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
#12 0x00007ffff7207922 in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::FastVector<unsigned long> const&, boost::optional<gtsam::FastMap<unsigned long, int> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, bool) () from /usr/local/lib/libgtsam.so.4
#13 0x000055555561add1 in mapOptimization::saveKeyFramesAndFactor (this=0x7fffffffbc80) at /usr/include/c++/9/bits/stl_vector.h:94
#14 0x000055555561e0be in mapOptimization::laserCloudInfoHandler (this=0x7fffffffbc80, msgIn=...) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:380
#15 0x00005555555e0b82 in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&>::operator() (a0=..., this=<optimized out>) at /usr/include/boost/function/function_template.hpp:677
#16 boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const>) (function_obj_ptr=..., a0=...) at /usr/include/boost/function/function_template.hpp:158
#17 0x00005555555e1dce in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::operator() (a0=..., this=0x555555744288) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:474
#18 ros::SubscriptionCallbackHelperT<boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&, void>::call (this=0x555555744280, params=...) at /opt/ros/noetic/include/ros/subscription_callback_helper.h:144
#19 0x00007ffff755d199 in ros::SubscriptionQueue::call() () from /opt/ros/noetic/lib/libroscpp.so
#20 0x00007ffff750b162 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/noetic/lib/libroscpp.so
#21 0x00007ffff750c873 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/noetic/lib/libroscpp.so
#22 0x00007ffff756002f in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/noetic/lib/libroscpp.so
#23 0x00007ffff754824f in ros::spin() () from /opt/ros/noetic/lib/libroscpp.so
#24 0x00005555555ae869 in main (argc=<optimized out>, argv=<optimized out>) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:2462

```

starry766, 2025-07-24:
> [@Noahcuptea](https://github.com/Noahcuptea) [@QingKe-UP](https://github.com/QingKe-UP) 我也遇到了这个问题，不知道有没有解决方案呢？现在我的报错是
> 
> ```
> double  free  or  corruption(out)    
>  Thread 1 "lio_sam_6axis_m" received signal SIGABRT, Aborted.   
>   __GI_raise (sig=sig@entry=6)  at ../sysdeps/unix/sysv/linux/raise.c:50  
>  50   ../sysdeps/unix/sysv/linux/raise.c:  没有那个文件或目  
> ```
> 
> 然后我还用gdb打印了栈信息，可是报错里面的内容是gtsam中优化函数的问题，我已经把gtsam重新安装了，安装的gtsam版本是4.0.0，Eigen是3.3.7版本。可是还没有解决问题。
> 
> ```
> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
> #1  0x00007ffff67be859 in __GI_abort () at abort.c:79
> #2  0x00007ffff6829266 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff6953298 "%s\n") at ../sysdeps/posix/libc_fatal.c:156
> #3  0x00007ffff68312fc in malloc_printerr (str=str@entry=0x7ffff6955670 "double free or corruption (out)") at malloc.c:5347
> #4  0x00007ffff6832fa0 in _int_free (av=0x7ffff6988b80 <main_arena>, p=0x55555575a930, have_lock=<optimized out>) at malloc.c:4314
> #5  0x00007ffff723285d in gtsam::NoiseModelFactor::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
> #6  0x00007ffff7233b3c in gtsam::(anonymous namespace)::_LinearizeOneFactor::operator()(tbb::blocked_range<unsigned long> const&) const () from /usr/local/lib/libgtsam.so.4
> #7  0x00007ffff7233f1d in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, gtsam::(anonymous namespace)::_LinearizeOneFactor, tbb::auto_partitioner const>::execute() () from /usr/local/lib/libgtsam.so.4
> #8  0x00007ffff6d06545 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
> #9  0x00007ffff6d0680f in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
> #10 0x00007ffff6d03b68 in ?? () from /lib/x86_64-linux-gnu/libtbb.so.2
> #11 0x00007ffff7235c9f in gtsam::NonlinearFactorGraph::linearize(gtsam::Values const&) const () from /usr/local/lib/libgtsam.so.4
> #12 0x00007ffff7207922 in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::FastVector<unsigned long> const&, boost::optional<gtsam::FastMap<unsigned long, int> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, bool) () from /usr/local/lib/libgtsam.so.4
> #13 0x000055555561add1 in mapOptimization::saveKeyFramesAndFactor (this=0x7fffffffbc80) at /usr/include/c++/9/bits/stl_vector.h:94
> #14 0x000055555561e0be in mapOptimization::laserCloudInfoHandler (this=0x7fffffffbc80, msgIn=...) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:380
> #15 0x00005555555e0b82 in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&>::operator() (a0=..., this=<optimized out>) at /usr/include/boost/function/function_template.hpp:677
> #16 boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const>) (function_obj_ptr=..., a0=...) at /usr/include/boost/function/function_template.hpp:158
> #17 0x00005555555e1dce in boost::function1<void, boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> >::operator() (a0=..., this=0x555555744288) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:474
> #18 ros::SubscriptionCallbackHelperT<boost::shared_ptr<lio_sam_6axis::cloud_info_<std::allocator<void> > const> const&, void>::call (this=0x555555744280, params=...) at /opt/ros/noetic/include/ros/subscription_callback_helper.h:144
> #19 0x00007ffff755d199 in ros::SubscriptionQueue::call() () from /opt/ros/noetic/lib/libroscpp.so
> #20 0x00007ffff750b162 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/noetic/lib/libroscpp.so
> #21 0x00007ffff750c873 in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/noetic/lib/libroscpp.so
> #22 0x00007ffff756002f in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) () from /opt/ros/noetic/lib/libroscpp.so
> #23 0x00007ffff754824f in ros::spin() () from /opt/ros/noetic/lib/libroscpp.so
> #24 0x00005555555ae869 in main (argc=<optimized out>, argv=<optimized out>) at /home/zhao/Codes/SLAM/LIO_SAM_history/src/LIO_SAM_6AIXS/LIO-SAM-6AXIS/src/mapOptmizationGps.cpp:2462
> ```

@zhao-zhibo 你好，请问你的问题解决了吗？我的也是出现在相同的地方。

allenthreee, 2025-12-18:
> 做好了，库之间一定要配套，我做了如下改动：
> 
> * ubuntu 20.04 -> 18.04 （ros neotic -> melodic）
> * 重装系统的分区 swag 与实际物理内存一致，系统里显示15左右gib=16gb
> * eigen 3.40 -> 3.3.4-4
> * 加了vtk 7.1.1
> * pcl 1.13.0 -> 1.8.1
> * opencv 4 -> opencv 3.4.19
> * 还装了显卡配套驱动
> 
> 推荐软件 timeshift，一个系统备份还原工具，方便restore至合适的节点

大佬，你做的docker还有吗，想是一下，搞了很久都没成功