hku-mars/FAST-LIVO #32: 试了一下自己录的数据集，效果还不错，请教一下作者关于相机的问题

Es1erda, 2022-12-15:
作者您好！ 感谢您这么优秀的开源工作，最近我用自己的录的数据集跑了一下效果很不错，视频链接如下，请作者指教一下。

【深圳大学FAST-LIVO建图测试】 https://www.bilibili.com/video/BV1rd4y1v7Fd/?share_source=copy_web&vd_source=c8d922f79df8ad8e31d0d55dd06658d9

以及有问题想请教一下，相机过曝的问题可以在mvs的驱动里面设置成自动曝光来解决吗？  以及有什么好的cmos相机推荐吗？我用的这一款只有130w像素，成像效果不是很好。

xuankuzcr, 2022-12-15:
感谢您对该工作的测试和兴趣！
关于曝光问题，在您测试的这种室外充满阳光的场景，我建议固定曝光在0.5ms左右，但打开自动增益GainAuto。或者在小范围内自动曝光，比如1ms以内自动曝光。
关于相机分辨率问题，我可以推荐一款500w像素的海康cmos，MV-CS050-10GC。我建议使用低分辨率图像（612*512）进行位姿估计，原图（2448*2048）用来上色。因为图像本身就是极强的非凸函数，高分辨率图像会增加它的非凸性。

Es1erda, 2022-12-18:
感谢回答！ 后续我根据您的建议再试试

Vickycyc, 2024-02-26:
@xuankuzcr 您好，我最近在嘗試您的專案，對於呈現出來的效果感到驚艷。為了讓品質提升，我想請問在使用低分辨率進行位姿估計並利用原圖分辨率上色這個部分，該如何改寫程式來實作？謝謝。

closed 2024-06-20 by xuankuzcr