hku-mars/FAST-LIVO #33: corrupted size vs. prev_size [laserMapping-2] process has died

ZZy129326999, 2022-12-19:
[image]

ZZy129326999, 2022-12-19:
I can well run FAST-LIO, R3LIVE. But I do not run FAST-LIVO in hku1.bag.

ZZy129326999, 2022-12-19:
`

        if (feats_undistort->empty() || (feats_undistort == nullptr))
        {
            // cout<<" No point!!!"<<endl;
            if (!fast_lio_is_ready)
            {
                first_lidar_time = LidarMeasures.lidar_beg_time;
                p_imu->first_lidar_time = first_lidar_time;
                LidarMeasures.measures.clear();
                cout<<"FAST-LIO not ready"<<endl;
                continue;
            }
        }
`
这个是我用 gdb 定位的错误，我的配置是Ubuntu20.04，其他包都没啥问题，也是最近两天重装的系统，应该不存在其他的干扰。

xuankuzcr, 2022-12-19:
gdb的终端提示可以发下么

ZZy129326999, 2022-12-19:
> ` src/FAST-LIVO/src/laserMapping.cpp
> 
> ```
>     if (feats_undistort->empty() || (feats_undistort == nullptr))
>     {
>         // cout<<" No point!!!"<<endl;
>         if (!fast_lio_is_ready)
>         {
>             first_lidar_time = LidarMeasures.lidar_beg_time;
>             p_imu->first_lidar_time = first_lidar_time;
>             LidarMeasures.measures.clear();
>             cout<<"FAST-LIO not ready"<<endl;
>             continue;
>         }
>     }
> ```

展开的gdb发到下边这个信息里了，LidarMeasures.measures.clear(); 这个是1306中断的代码。感谢您啦！！

xuankuzcr, 2022-12-20:
用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures);` 替换掉`LidarMeasures.measures.clear();`试试

ZZy129326999, 2022-12-20:
> std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)

0x0}, step = {p = 0x0, buf = {0, 0}}}})
    at /usr/include/c++/9/bits/deque.tcc:870
#23 0x00005555555b7d73 in std::deque<MeasureGroup, std::allocator<MeasureGroup> >::_M_destroy_data(std::_Deque_iterator<MeasureGroup, MeasureGroup&, MeasureGrou--Type <RET> for more, q to quit, c to continue without paging--
p*>, std::_Deque_iterator<MeasureGroup, MeasureGroup&, MeasureGroup*>, std::allocator<MeasureGroup> const&) (__last=..., __first=..., this=0x7fffffff8f80)
    at /usr/include/c++/9/bits/stl_deque.h:167
#24 std::deque<MeasureGroup, std::allocator<MeasureGroup> >::~deque()
    (this=0x7fffffff8f80, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/stl_deque.h:1072
#25 0x000055555558cf7a in main(int, char**)
    (argc=<optimized out>, argv=<optimized out>)
    at /usr/include/c++/9/bits/move.h:182
(gdb) 
(gdb) up 25
#25 0x000055555558cf7a in main (argc=<optimized out>, argv=<optimized out>)
    at /usr/include/c++/9/bits/move.h:182
182	    swap(_Tp& __a, _Tp& __b)

ZZy129326999, 2022-12-20:
> 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试

感谢您的指教！！但是现在gdb定位不到具体位置了，感觉这个错误有些摸不到头绪。

ZZy129326999, 2022-12-20:
> 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试

        {img_offset_time = 0.0017692896071821451, imu = std::deque with -32 elements, img = {flags = 0, dims = 0, rows = 536870912, cols = 1067186637, data = 0x3fa772c8c0000000 <error: Cannot access memory at address 0x3fa772c8c0000000>, datastart = 0x3fefaeaec0000000 <error: Cannot access memory at address 0x3fefaeaec0000000>, dataend = 0x0, datalimit = 0x0, allocator = 0x0, u = 0x0, size = {p = 0x0}, step = {p = 0x0, buf = {0, 0}}}}) at /usr/include/c++/9/bits/stl_deque.h:2117
#25 0x00005555559d9dc0 in std::deque<MeasureGroup, std::allocator<MeasureGroup> >::~deque() (this=0x7fffffff9770, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/stl_deque.h:1072
#26 0x00005555559c98f5 in main(int, char**) (argc=1, argv=0x7fffffffd698)
    at /home/zzy/work/catkin_ws/fast_livo_ws/src/FAST-LIVO/src/laserMapping.cpp:1306
(gdb) 
(gdb) 
(gdb) up 26
#26 0x00005555559c98f5 in main (argc=1, argv=0x7fffffffd698)
    at /home/zzy/work/catkin_ws/fast_livo_ws/src/FAST-LIVO/src/laserMapping.cpp:1306
1306	                std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures);

这次我把rosbag play 停掉了，又有了。刚才没关rosbag就没定位到好像。 我顺便改成-O0了。麻烦您啦！！

ZZy129326999, 2022-12-20:
> 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试

[image]

ZZy129326999, 2022-12-20:
> 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试
我尝试输出了deque的长度是1。

crowningwang, 2023-01-29:
> > 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试
> > 我尝试输出了deque的长度是1。

Hello, have you solved this problem? I have the same question.

sebastiantorres1, 2023-02-01:
I also have the same question, but in my case I'm using a Jetson Orin and when i use the command
$ roslaunch fast_livo mapping_avia.launch 
the launch shows me this 

[image]

and the rviz doesn't show me the hku2.bag file.
I made the same test in my computer and it works fine, do you know how can i solve this problem?

xuankuzcr, 2023-02-06:
> > 用`std::deque<struct MeasureGroup> ().swap(LidarMeasures.measures)` 替换`LidarMeasures.measures.clear()`试试
> > 我尝试输出了deque的长度是1。

这个长度是1是没问题的，是别的问题，主要我这边无法复现你的问题，你解决这个问题了么？

closed 2023-02-08 by xuankuzcr

widmer2001, 2023-11-23:
你好 我也遇到了同样的问题，请问你解决了嘛？