hku-mars/FAST-LIVO #36: #define USE_IKFOM  后的 IKFoM_toolkit库报错

Michael-Zuo, 2023-01-10:
感谢优秀的工作，大部分情况都运行良好。

对于部分数据集，我在测试时发现初始化时FAST-LIVO总是跑飞, 但FAST-LIO2可以正常运行，于是我试了试在common_lib.h 中去掉了#define USE_IKFOM的注释，但出现了很多 IKFoM_toolkit库报错及其他报错。
如：
`/home/zuo/catkin_ws/src/R3LIVE/FAST-LIVO/include/IKFoM_toolkit/esekfom/../mtk/build_manifold.hpp:102:47: error: request for member ‘oplus’ in ‘((state_ikfom*)this)->state_ikfom::grav’, which is of non-class type ‘int’
 #define MTK_OPLUS(               type, id) id.oplus(MTK::subvector_(__vec, &self::id), __scale);
                                               ^
/home/zuo/catkin_ws/src/R3LIVE/FAST-LIVO/include/IKFoM_toolkit/esekfom/../mtk/build_manifold.hpp:102:47: note: in definition of macro ‘MTK_OPLUS’
 #define MTK_OPLUS(               type, id) id.oplus(MTK::subvector_(__vec, &self::id), __scale);
`
`/home/zuo/catkin_ws/src/R3LIVE/FAST-LIVO/include/IKFoM_toolkit/esekfom/../mtk/build_manifold.hpp:102:85: error: no matching function for call to ‘subvector_(const MTK::vectview<const double, 0>&, int state_ikfom::*)’
 #define MTK_OPLUS(               type, id) id.oplus(MTK::subvector_(__vec, &self::id), __scale);
`

`/FAST-LIVO/src/IMU_Processing.cpp:256:31:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: INVALID_MATRIX_PRODUCT
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
`

请问如何处理此类报错，或者如何正确使用#define USE_IKFOM ，是否还需要其他设置。
或对于初始化时FAST-LIVO部分情况跑飞的情况如何处理？是否需要先原地静止再移动？
十分感谢

[image]

xuankuzcr, 2023-01-12:
您好，感谢您对我们工作的兴趣～
1. 目前不支持`#define USE_IKFOM`模式；
2. 初始化时FAST-LIVO部分需要原地静止5s以上再移动，否则容易因cov_acc，cov_gyr以及gravity设置不合理导致ekf不收敛；
3. 如果对于部分数据集，发现不能work，核查外参和时差。你也可以把你的bag和参数文件发给我，我来测试一下。

wlwsjl, 2023-03-23:
你好，感谢你们开源的LVIO。
我想测试HILTI 2022 Dataset，选取的Sequence是Exp04。外参做了相应的适配，IMU noise参考的是如下文件：
https://github.com/hku-mars/FAST_LIO/blob/main/config/velodyne.yaml
但是我遇到类似的问题，FAST-LIVO总是跑飞（无论是否enable_img)，但FAST-LIO2可以正常运行。
我尝试了修改IMU noise的方法但是没有成功。
请问你们是如何解决这个问题的？
是否可以分享一下支持该数据集的config file?