hku-mars/FAST-LIVO #38: Use-after-free bug in img_cbk()

kekeliu-whu, 2023-01-28:
Hi, I find a bug by [AddressSanitizer](https://github.com/google/sanitizers/wiki/AddressSanitizer) when testing our dataset:
```cpp
cv::Mat getImageFromMsg(const sensor_msgs::ImageConstPtr& img_msg) {
  cv::Mat img;
  img = cv_bridge::toCvShare(img_msg, "bgr8")->image;
  return img;
}
```
It can be seen from the description of `toCvShare()` that `img_msg->data` and `img.data` may share the same memory, which leads to a use-after-free bug. To be more specifically, `img_msg->data` will be released after `getImageFromMsg()` but the return value `img` may still be used after that.

xuankuzcr, 2023-02-06:
Great discovery and a very practical tool! Additionally, could you share your configuration operation of [AddressSanitizer](https://github.com/google/sanitizers/wiki/AddressSanitizer) in the project?

kekeliu-whu, 2023-02-07:
Tool usage reference:
https://github.com/google/sanitizers/wiki/AddressSanitizer

Usage:
1. Change the cmakelists.txt as below:
```diff
4,8c4
< SET(CMAKE_BUILD_TYPE "Debug")
<
< ADD_COMPILE_OPTIONS(-std=c++14 )
< ADD_COMPILE_OPTIONS(-std=c++14 )
< set( CMAKE_CXX_FLAGS "-std=c++14 -O3" )
---
> SET(CMAKE_BUILD_TYPE "")
16c12
< set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14 -pthread -std=c++0x -std=c++14 -fexceptions")
---
> set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fsanitize=address -O1 -fno-omit-frame-pointer -g -fexceptions")
```
2. Build fast-livo with `cmake .. && make`
3. Run fast-livo to detect memory issue.

Following the steps above, maybe more issues could be found :)

I have made a pull request to fix the bug above: https://github.com/hku-mars/FAST-LIVO/pull/39. Hope you can merge it!

xuankuzcr, 2023-02-07:
Thanks for your sharing. I have merged the PR.

commit a8918c23 (2023-02-07): Merge pull request #39 from kekeliu-whu/fix-use-after-free

Fix #38
closed 2023-02-07 by xuankuzcr