hku-mars/FAST-LIVO #40: [laserMapping-1] process has died [pid 31651, exit code -11

crowningwang, 2023-01-28:
尊敬的港大团队，您好
在编译完成之后运行roslaunch fast_livo mapping_avia.launch后laserMapping中断，并且播放数据集rviz没有显示，请问应该如何解决这个问题，我的配置如下：
Ubuntu->18.04
PCL->1.9.1
Eigen->3.4.0
OpenCV->4.2.0
[image]

crowningwang, 2023-01-28:
[image]

closed 2023-02-01 by crowningwang

IaroslavS, 2023-02-01:
> [image]

How did you solve this problem ?

xuankuzcr, 2023-02-06:
你用gdb看一下哪行报错了，opencv推荐3.x的版本

justleap, 2023-02-24:
请问这个问题是怎么解决的，遇到同样的问题，rviz有时不显示有时可以，跑的时候会出现进程崩溃

autumntang, 2025-03-20:
Error: RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error