hku-mars/FAST-LIVO #41: [laserMapping-1] process has died [pid 11250, exit code -11

IaroslavS, 2023-02-03:
If anybody faces this kind of error (look at the screen shot), please  pay attention to string 
```
printf("[ LIO ]: Using multi-processor, used core number: %s.\n", MP_PROC_NUM);
```
in laserMapping.cpp

Resolve the bug by changing the above string to the following:
```
printf("[ LIO ]: Using multi-processor, used core number: %d.\n", MP_PROC_NUM);
```

[image]

sebastiantorres1, 2023-02-03:
I had made a research in the lasermappin.cpp code and apparently the process stops working in this part of the code: 

` if(!vk::camera_loader::loadFromRosNs("laserMapping", lidar_selector->cam))`
`      throw std::runtime_error("Camera model not correctly specified."); `

Have you faced this problem?

I tried your solution and it doesn't fix my problem.

IaroslavS, 2023-02-04:
> vk::camera_loader::loadFromRosNs("laserMapping", lidar_selector->cam

Yes, I faced similiar problem, check out config file that you use when launching, for example this config file
https://github.com/hku-mars/FAST-LIVO/blob/main/config/camera_NTU_VIRAL.yaml#L1
has field `cam_model` which should be one of the set {Ocam, Pinhole, ATAN}

Check or edit this file 
https://github.com/uzh-rpg/rpg_vikit/blob/master/vikit_ros/include/vikit/camera_loader.h
that repository (https://github.com/uzh-rpg/rpg_vikit/) is a submodule for fast-livo

sebastiantorres1, 2023-02-06:
Apparently i just have to change the camera model "Pinhole" to "ATAN" in the configuration, but now it only shows the camera and the trajectory, but not the lidar odometry in the rviz  like this image shows

[image]

How did you solve this?

xuankuzcr, 2023-02-08:
As for our provided bags, you should use `Pinhole` camera model. While running your own collected dataset, choose the camera model according to your device.

closed 2023-02-08 by xuankuzcr