hku-mars/FAST-LIVO #42: question about LidarSelector::addObservation()

zzwu29, 2023-02-14:
Dear Authors:
Thank you for contributing elegant programs to open-source SLAM, which helped me a lot in my learning. During this process, I have a question about **LidarSelector::addObservation()** 
https://github.com/hku-mars/FAST-LIVO/blob/a8918c23dc4dba7d98fa723a9fa0fc2cbdf4d977/src/lidar_selection.cpp#L951-L952
From my perspective, it aims to calculate the change of rotation and translation between frames, but should the threshold for angular change be set to 10 degrees instead of 10 radians?
Wish you all the best!

xuankuzcr, 2023-02-15:
I have changed it in the new version of the code, but forgot to update the old version. The experience value of **delta_theta** can be set to 0.3.

zzwu29, 2023-05-04:
Thank you for your kind advice!