hku-mars/FAST-LIVO #43: question about ImuProcess::UndistortPcl()

zzwu29, 2023-02-14:
Dear Authors:
Thank you for contributing elegant programs to open-source SLAM, which helped me a lot in my learning. During this process, I have a question about **ImuProcess::UndistortPcl()**
https://github.com/hku-mars/FAST-LIVO/blob/a8918c23dc4dba7d98fa723a9fa0fc2cbdf4d977/src/IMU_Processing.cpp#L808-L811
From my point of view, the fast-livo code corresponds to the following formula
[image]
Does it use the extrinsic rotation of Lidar and IMU? Please help check whether the following formula is correct
[image]
R is the direction cosine matrix, p is the translation, L is lidar frame, G is global frame, and b is IMU frame.
Wish you all the best!

xuankuzcr, 2023-02-15:
In the `T_ei`, `e` is in the IMU frame, and `i` is in the LiDAR frame. Therefore, the extrinsic LiDAR frame w.r.t. IMU frame is used only once.

zzwu29, 2023-02-15:
您好，您的邮件我已经收到，会尽快阅读并回复！
                                                                               zzWu

zzwu29, 2023-02-15:
Please forgive my trouble, the translation of lidar w.r.t. IMU is Lid_ offset_ to_ IMU, while the rotation is expressed as Lid_ rot_ to_ IMU, right? When will the extrinsic rotation parameter of lidar w.r.t. IMU be used?

closed 2023-10-09 by zzwu29