hku-mars/FAST-LIVO #45: 请问一下关于相机参数设置方面的 问题

Ningningwang11, 2023-02-15:
首先我使用hku1.bag的包是可以正常运行fast-livo
但是我换了自己的设备：主机（jetson agx xavier）、livox-mid70、imu(wheeltec)、ZED2。
这是我遇到的问题
[image]
为了排除雷达和imu是不是跑的通fast-lio我把img_enable设置为0后又跑了一下程序，发现rviz虽然没有点云图，但是看样子应该是跑的了fast-lio的，
[image]
。所以我觉得可能是相机标定信息什么的设置不对。
基于相机的出厂标定信息修改了下面的部分
[image]

我对这一块刚入门，想请教一下，我都需要改哪些参数，才可以像跑hku1.bag一样基于我的设备跑通fast-livo。 非常感谢！

Ningningwang11, 2023-02-15:
修改了

Ningningwang11, 2023-02-15:
cam_width:   640 #672  # 640
cam_height:  360 #376    #512
原来是图片分辨率需要调

closed 2023-02-15 by Ningningwang11

Aokiji996, 2023-04-05:
我的设备跟你几乎一样，我用的zed内置的imu，能看看你的config文件吗

Ningningwang11, 2023-04-05:
现在单跑fastlio2感觉效果很不错，但是跑fastlivo后虽然能跑，效果有点差。但是雷达-imu，相机-雷达都标定了，还是跑的远没有他原版数据包效果好。甚至说效果有些差。明天我去翻一下那个文件给你，感觉仅供参考了，如果明天中午12点前我没有发的话，请在提醒我一下。

发自我的iPhone

Aokiji996, 2023-04-05:
太感谢了

Ningningwang11, 2023-04-06:
zed内参 我用的是原厂数据，雷达-imu标定，相机-雷达标定的包都可以在github上找到。
我最终跑的效果有点差，如果您最终跑通的效果比较好，点云上色效果比较准确，定位比较准确的话，还请回复我您觉得的关键操作细节，多谢。

898074006
***@***.***

Aokiji996, 2023-04-06:
我的mid70获取数据之后的时间戳似乎有些问题，我怀疑这是导致我不能跑通的原因，但我不知道要怎么才能正确的设置这个时间戳
[image]

Ningningwang11, 2023-04-06:
我的笔记里有ptcd什么的你查一下

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Aokiji996 ***@***.***&gt;
发送时间: 2023年4月6日 19:58
收件人: hku-mars/FAST-LIVO ***@***.***&gt;
抄送: Ningningwang11 ***@***.***&gt;, State change ***@***.***&gt;
主题: Re: [hku-mars/FAST-LIVO] 请问一下关于相机参数设置方面的 问题 (Issue #45)

Ningningwang11, 2023-04-06:
加个微信，
QQ898974006

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Aokiji996 ***@***.***&gt;
发送时间: 2023年4月6日 19:58
收件人: hku-mars/FAST-LIVO ***@***.***&gt;
抄送: Ningningwang11 ***@***.***&gt;, State change ***@***.***&gt;
主题: Re: [hku-mars/FAST-LIVO] 请问一下关于相机参数设置方面的 问题 (Issue #45)

Ningningwang11, 2023-04-06:
尬住了，我QQ 打错了
898074006

发自我的iPhone

 
我已经添加了你的QQ，收到了吗