hku-mars/FAST-LIVO #47: 【求助】定义it数组时遇到段错误

Noahcuptea, 2023-02-22:
尊敬的港大团队您们好，

设备都调试硬同步好了，程序跑起来开始报错。
可能是从点云中筛选出与图像匹配的点，将这些点的坐标存入it中的时候出了一些错误。
我挠破头都不知道怎么整了，下面是是rosbag和配置文件的下载链接，还有gdb信息您过目：

[百度网盘下载](https://pan.baidu.com/share/init?surl=nthjgtVyyc99i3HrhnVRmA&pwd=rbxp)

**开始GDB:**

```
Thread 1 "fastlivo_mappin" received signal SIGSEGV, Segmentation fault.
0x00007ffff03df268 in lidar_selection::LidarSelector::addFromSparseMap (
    this=this@entry=0x55555c676f50, img=..., pg=...)
    at /catkin_ws/src/FAST-LIVO/src/lidar_selection.cpp:376
376	    float it[height*width] = {0.0};
```

(gdb) bt
```
#0  0x00007ffff03df268 in lidar_selection::LidarSelector::addFromSparseMap(cv::Mat, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZINormal> >) (this=this@entry=0x55555c676f50, img=..., pg=...)
    at /catkin_ws/src/FAST-LIVO/src/lidar_selection.cpp:376
#1  0x00007ffff03e4893 in lidar_selection::LidarSelector::detect(cv::Mat, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZINormal> >) (this=0x55555c676f50, img=..., pg=...) at /catkin_ws/src/FAST-LIVO/src/lidar_selection.cpp:1065
#2  0x000055555557fd56 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /catkin_ws/src/FAST-LIVO/src/laserMapping.cpp:1344
```

(gdb) print img
```
$1 = {flags = 1124024320, dims = 2, rows = 2048, cols = 2048, 
  data = 0x5555607f71c0 "\t\r\f\f\v\f\004\006\b\f\b\006\006\n\n\f\v\v\r\r\v\016\n\v\f\t\003\v\f\b\006\b\a\v\n\t\r\v\002\006\f\f\016\f\t\v\020\f\n\b\a\n\f\b\a\n\016\r\t\r\016\v\v\026\033\030\023\023\022\022\026\025\025\025\017\017\026\024\027\016\003\006\a\003\001\v\017\006\005\r\f\n\v\b\a\a\r\f\n\t\004\002\003\001\003\t\n\006\b\a\b\r\r\016\017\017\021\017\v\t\017\016\n\n\v\016\020\021\021\024\024\017\r\024\025\020\023\020\016\020\024\022\016\t\a\f\017\n\n\v\f\n\n\v\006\t\v\016\t\002\b\a\005\a\b\r\r\v\r\n\b\n\v\t\n\006\a\v\016\f\005\n\r\t\002\b\b\t\b\v\004\t\020\t\f\020\016\t\n\r"..., 
  datastart = 0x5555607f71c0 "\t\r\f\f\v\f\004\006\b\f\b\006\006\n\n\f\v\v\r\r\v\016\n\v\f\t\003\v\f\b\006\b\a\v\n\t\r\v\002\006\f\f\016\f\t\v\020\f\n\b\a\n\f\b\a\n\016\r\t\r\016\v\v\026\033\030\023\023\022\022\026\025\025\025\017\017\026\024\027\016\003\006\a\003\001\v\017\006\005\r\f\n\v\b\a\a\r\f\n\t\004\002\003\001\003\t\n\006\b\a\b\r\r\016\017\017\021\017\v\t\017\016\n\n\v\016\020\021\021\024\024\017\r\024\025\020\023\020\016\020\024\022\016\t\a\f\017\n\n\v\f\n\n\v\006\t\v\016\t\002\b\a\005\a\b\r\r\v\r\n\b\n\v\t\n\006\a\v\016\f\005\n\r\t\002\b\b\t\b\v\004\t\020\t\f\020\016\t\n\r"..., 
  dataend = 0x555560bf71c0 "\023#0\024$1\025%\241e;", 
  datalimit = 0x555560bf71c0 "\023#0\024$1\025%\241e;", allocator = 0x0, 
  u = 0x55555c679f00, size = {p = 0x7fffffff4988}, step = {p = 0x7fffffff49d0, 

    buf = {2048, 1}}}
```
(gdb) print pg
`$2 = {px = 0x55555c2b9840, pn = {pi_ = 0x55555c2b98e0}}`

---

**以下是录制Rosbag时候camera和livo节点的报错，关掉rosbag record报错就没了**

Camera 节点
`Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).`

Livo节点
```
[ERROR] [1677054307.767761744]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[ERROR] [1677054307.767983097]: OpenCV(3.4.19) /opt/opencv_build/opencv/modules/imgcodecs/src/loadsave.cpp:1000: error: (-2:Unspecified error) in function 'bool cv::imencode(const cv::String&, cv::InputArray, std::vector<unsigned char>&, const std::vector<int>&)'
> Encoding 'params' must be key-value pairs:
>     '(params.size() & 1) == 0'
> where
>     'params.size()' is 3
```

xuankuzcr, 2023-02-26:
这好像是M2DGR的数据数采设备吧。没看懂你的问题，你是边record边在跑LIVO么？https://answers.ros.org/question/251403/image_transport-rosbag-issue/

Noahcuptea, 2023-02-27:
> 这好像是M2DGR的数采设备吧。没看懂你的问题，你是边record边在跑LIVO么？https://answers.ros.org/question/251403/image_transport-rosbag-issue/

谢谢您的回复，用的是livox-mid-70, 外置imu，海康camera，所有硬件同步已完成。

我问了两个问题：
第一个是关于Camera数据接入后开始报错，我提供了gdb调试的报错信息和相关的rosbag.
第二个是”边record边在跑LIVO“的情况下，会出现更多的报错，仅当参考。
咱们不管第二个，只说第一个问题，它的错误链条是：
laserMapping.cpp:1344 ---> lidar_selection.cpp:1065 ---> lidar_selection.cpp:376
也就是在 传递（img=..., pg=...）的时候出错了，请问我现在改怎么排查？
您方便看下rosbag和配置文件指点一下么？

在下从最早的lio1开始跟随您实验室的工作，
每次复现成功都会上传建筑设计行业落地应用的一些案例，
希望这次也有机会为这个项目提供一些数据，再次万分感激您宝贵的时间。

xuankuzcr, 2023-02-27:
[image]
你这个内参和畸变有大问题啊，cx怎么可能比图像宽度还大啊

Cc19245, 2023-03-04:
您好，请问这个问题目前解决了吗？我也遇到了同样的问题：
```
[New Thread 0x7fff7e7fc700 (LWP 19081)]
[New Thread 0x7fff7dffb700 (LWP 19082)]
[ LIO ]: time: fov_check: 0.000000 fov_check and readd: 0.002374 match: 0.000000 solve: 0.000000  ICP: 0.000474  map incre: 0.004058 total: 0.006905 icp: 0.000416 construct H: 0.000000.
[ INFO ]: get img at time: 1579245924.295512.
[ INFO ]: get point cloud at time: 1579245924.243978 and size: 4291.
[ VIO ]: Raw feature num: 4317.
[ VIO ]: Add 589 3D points.
[ VIO ]: time: addFromSparseMap: 0.000000 addSparseMap: 0.002141 ComputeJ: 0.000000 addObservation: 0.000000 total time: 0.002142 ave_total: 0.002142.
[ INFO ]: get img at time: 1579245924.345247.
[ VIO ]: Raw feature num: 4317.

Thread 1 "fastlivo_mappin" received signal SIGSEGV, Segmentation fault.
0x00007ffff2aa9088 in lidar_selection::LidarSelector::addFromSparseMap (this=this@entry=0x55555c7b40d0, img=..., 
    pg=...) at /home/cc/fast-livo-ws/src/FAST-LIVO/src/lidar_selection.cpp:376
---Type <return> to continue, or q <return> to quit---
376         float it[height*width] = {0.0};
(gdb)
```

> [image] 你这个内参和畸变有大问题啊，cx怎么可能比图像宽度还大啊

Cc19245, 2023-03-04:
我把报错的地方
```
float it[height * width] = {0.0}; 
```
修改成

```
std::vector<float> it(height*width, 0);
```
问题解决了。

但是不知道为什么原来的写法在hku数据集上没有问题，在我自己的数据上就会报错呢？我的图像是1920*1200，和这个有关吗？

Noahcuptea, 2023-03-09:
> 我把报错的地方
> 
> ```
> float it[height * width] = {0.0}; 
> ```
> 
> 修改成
> 
> ```
> std::vector<float> it(height*width, 0);
> ```
> 
> 问题解决了。
> 
> 但是不知道为什么原来的写法在hku数据集上没有问题，在我自己的数据上就会报错呢？我的图像是1920*1200，和这个有关吗？

不好意思才回复，刚刚身体康复回到工作室。
是的，报错跟您的图像有关，参数包括：
1. cam_width 和 cam_height 表示图像的宽度和高度，
2. cam_fx 和 cam_fy 是相机的焦距，
3. cam_cx 和 cam_cy 是相机光心在图像平面上的坐标，
4. cam_d0、cam_d1、cam_d2 和 cam_d3 是相机的畸变参数。

据我所知，这些都在相机和激光雷达的**联合标定过程**中可以得到，您可能需要参考如下内容：
1. [Livox Lidar+海康Camera实时生成彩色点云](https://blog.csdn.net/weixin_41965898/article/details/117447600)
2. [lidar_camera_calib](https://github.com/hku-mars/livox_camera_calib)

我还没解决报错，祝您复现顺利哈 ：）