hku-mars/FAST-LIVO #50: when running eee_03.bag of NTU_VIRAL,it report an error about [laserMapping-2] process has died...

hsiaochao, 2023-03-07:
when running eee_03.bag of NTU_VIRAL,it report an error about [laserMapping-2] process has died...
[laserMapping-2] process has died [pid 3278, exit code -11, cmd /root/catkin_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping __log:=/root/.ros/log/552d2f00-bccf-11ed-937d-6c24081e023e/laserMapping-2.log].
log file: /root/.ros/log/552d2f00-bccf-11ed-937d-6c24081e023e/laserMapping-2*.log
How to solve it？

xuankuzcr, 2023-03-13:
Hi，can you give me some gdb debug info?

seajayshore, 2023-03-14:
eee_03.bag works in my Fast-LIVO setup. 
From your error you **might** be having the same issue I did originally - see this #53 (Section "Crash When Running My Own Data:")

Also as @xuankuzcr suggests gdb is very useful. See the same issue I just linked if you have never used it before.

closed 2023-10-24 by xuankuzcr

Ab-Tx, 2025-03-14:
The solution for this issue is to update the c++14 to c++17 in the "CMakeLists.txt" of FAST-LIVO.

For anyone having issues, you can replace these lines in the CMakeLists.txt:
```
GNU nano 6.2                                                 CMakeLists.txt                                                          
cmake_minimum_required(VERSION 3.11.4)
project(fast_livo)

SET(CMAKE_BUILD_TYPE "Debug")

ADD_COMPILE_OPTIONS(-std=c++17 )
ADD_COMPILE_OPTIONS(-std=c++17 )
set( CMAKE_CXX_FLAGS "-std=c++17 -O3" ) 

add_definitions(-DROOT_DIR=\"${CMAKE_CURRENT_SOURCE_DIR}/\")

set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fexceptions" )
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17 -march=native -pthread -std=c++0x -std=c++17 -fexceptions")
SET(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${PROJECT_SOURCE_DIR}/cmake_modules/")
```
I also applied the modification suggested in #53 (regarding the float variable), but I don't think was is involved in the algorithm crash.

# Why? 
[Eigen.](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html)

# How?
Using GDB as instructed in #53, and the dockerfile provided in #136 .
This [quick guide for ROS & GDB is also useful.](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB)

# Output
Here is the output I got in RVIZ.
[image]