hku-mars/FAST-LIVO #54: Help to Confirm Hardware Synchronisation Requirements

seajayshore, 2023-03-14:
This question is similar to #21, #23 and possibly others. I want to make Fast-LIVO work well with my custom data but I'm not sure what the precise synchronization requirements are - the word "synchronization" can be used differently so I hope to make it 100% clear. 🤞

### Details:
To ask my question clearly I will use the terms "event-synchronized" and "time-synchronized" as follows - hopefully this makes sense:

**Event-synchronized** = 2+ events happen at the exact same moment
  - E.g. 2 cameras are triggered to capture 2 images simultanously but they **don't necessarily** share the same clock so the 2 images won't have the same timestamp even though they were taken at the same moment

**Time-synchronized** = 2+ sensors share the same clock / time-domain
  - E.g. 2 cameras have a PTP or PPS connection so their internal timestamps can be kept in-sync with eachother but they **don't necessarily** capture images at the exact same moment 

Often we need *both* good event-sync & time-sync but often only time-sync is required.
So basically this is my question - what kind of sync do we need for Fast-LIVO to work well?

### Your Datasets:
In your own data (e.g. hku2.bag) the images & lidar messages clearly have the **same timstamp** and **you are triggering them simultanously** using the microcontroller. So in my words this data is *both* precisely "event-synchronized" and "time-synchronized":
[image]

### NTU_VIRAL Datasets:
However in the NTU_VIRAL data (e.g. eee_03.bag) the images & lidar messages occur at different times but *are* precisly referenced to a common clock of some kind. So in my words this data is precisely "time-syncronized" but **not** "event-syncronized": 
[image]

### My Datasets:
My data is more like NTU_VIRAL as my images are 15Hz and the Lidar (Mid-70) is 10Hz so they don't natually align. My data is not working very well in Fast-LIVO at the moment - obvisouly this could be for many reasons - but I want to first make 100% sure about the syncronization aspect otherwise other improvements are pointless. 
[image]

The only difference I noticed between the config file for your data & NTU_VIRAL is that the `ncc_en` & `ncc_thre` paramaters are set - is this related? If so can you help explain a little what to do?

FPSychotic, 2023-03-15:
Until they answer you, my experience is,you need basically global time
synchro more than even sychro, small differences in even synchro doesn't
look affect the quality in any significant way, at least at human walk
speed and maps if moderate size, i.e. a few handred meters.
 I think that a moderate event difference is corrected by the back
propagation feature. I'm not programmer,just integrator, but I just
suppose, if the not perfect event synchro would affect me, it won't happens
before I run out of memory.

You can rise the Hz of your lidar to match you frame rate, the lidar won't
rise the points speed, still will work at 10hz internally, but will correct
de data with the Hz you want up to 100hz I think is the limit.

Global shutter or rolling shutter looks don't make difference if you use
30hz or more at least, for sure is better global shutter , but at 30hz it
looks the software is working good, tested with two different streams from
a bag, probably at 15Hz is necessary global shutter.

That is one of the three or four mayor things that's affect the mapping,
you need be sure you have at least 30hz ,global shutter or not, as it is
used for VIO, not only for point cloud colourization.

The second one is is the colour  resolution/computing power/free memory
relation, you need be sure there is not jumps or delays, and the map
resolution will affect in that if you have not a very beefy computer
specially at 30fps or more and less of 0.05m or less, for 0.05 you will
need an i7 8core  8  threads 32gb ram and nvme to have reliable FPS to
don't suffer jumps or delay.

The other thing that affects massively is the IMU Gravity settings, you
will need a good gravity setup and delay.

The one most important is the calibration of extrinsic camera/IMU and
camera intrinsics. No only alignment won't be good, a bad calibration will
rise the computing requeriment, not sure if by the covariance calculus, or
why, but the CPU use will rice notably  that will affect the second point.

Also record bag, and save mapping can affects the stability, make jumps in
frame rate and time stamps and mess everything, even in a seven, if you
Dave bag, better map in post, if you map realtime, better don't make bag,
at least in a i7 8 threads, in 16 threads or i9 maybe is different.

You need process the software and data with easy.
A example of real time with 30fps rolling shutter, 30hz lidar, 0.025
resolution and only NTP time synchro in a i7 8cores 16 ram nvme.

https://youtu.be/NRawkwDvlig

https://youtu.be/NRawkwDvlig

xuankuzcr, 2023-12-06:
Hello, we have updated our synchronization scheme. Compared with the synchronization method in the LIVO paper, this new approach also sends a 1 Hz fake GPRMC message generated by the MCU to the Livox Avia LiDAR through the serial port. This ensures that the LiDAR scan and the camera frame are triggered at the same moment. Subsequently, the LiDAR scan header timestamp is assigned to the camera messages through shared memory (in the camera driver). This is why the camera and LiDAR frame header timestamps in our open-sourced data on GitHub are exactly the same; each frame of data from the camera and LiDAR is collected at the same moment (not considering exposure time and circuit delay). In addition, we have resolved the issue of timestamp rollbacks for the LiDAR, IMU, and LiDAR points, and compensated for this time in the Livox ROS driver.

closed 2024-06-20 by xuankuzcr