hku-mars/FAST-LIVO #56: 跑自己数据集时无效，while循环卡在if(!sync_packages(LidarMeasures))

ZRS0326, 2023-03-21:
问题1,我自己的数据集是MID360雷达录制的，点云格式是livox_ros_driver2::CustomMsg，实际上与livox_ros_driver::CustomMsg一致。日志输出已获取到点云，不过LIO模块不执行，经测试卡在sync_packages(LidarMeasures)中的
        meas.lidar = lidar_buffer.front(); // push the firsrt lidar topic
        if(meas.lidar->points.size() <= 1)
        {
            mtx_buffer.lock();
            if (img_buffer.size()>0) // temp method, ignore img topic when no lidar points, keep sync
            {
                lidar_buffer.pop_front();
                img_buffer.pop_front();
            }
            mtx_buffer.unlock();
            sig_buffer.notify_all();
            // ROS_ERROR("out sync");
            cout<<"A3"<<endl;
            **return false;**   卡在这一步
        }
请问是数据集的关系吗？想要寻求解决方式。
问题2，MID360雷达也是内置IMU，但在我启动后发现IMU初始化似乎没运行。
void ImuProcess::IMU_init(const MeasureGroup &meas, StatesGroup &state_inout, int &N)
{
  /** 1. initializing the gravity, gyro bias, acc and gyro covariance
   ** 2. normalize the acceleration measurenments to unit gravity **/
  ROS_INFO("IMU Initializing: %.1f %%", double(N) / MAX_INI_COUNT * 100);  **这一步没有执行**
  V3D cur_acc, cur_gyr;

SunZezhou, 2023-03-23:
I noticed the problem occurred with the tag of Mid360 points. 

It works well for me after commenting out the following line, but I don't know what this line means.

https://github.com/hku-mars/FAST-LIVO/blob/a8918c23dc4dba7d98fa723a9fa0fc2cbdf4d977/src/preprocess.cpp#L139

[image]

xuankuzcr, 2023-03-28:
> 问题1,我自己的数据集是MID360雷达录制的，点云格式是livox_ros_driver2::CustomMsg，实际上与livox_ros_driver::CustomMsg一致。日志输出已获取到点云，不过LIO模块不执行，经测试卡在sync_packages(LidarMeasures)中的 meas.lidar = lidar_buffer.front(); // push the [...]

核对下IMU和LiDAR之间的时间戳吧，最好打印出来。

xuankuzcr, 2023-03-28:
> I noticed the problem occurred with the tag of Mid360 points.
> 
> It works well for me after commenting out the following line, but I don't know what this line means.
> 
> https://github.com/hku-mars/FAST-LIVO/blob/a8918c23dc4dba7d98fa723a9fa0fc2cbdf4d977/src/preprocess.cpp#L139
> 
> [image]

这是通过mid360的tag判断回波的，注释掉试试？

closed 2023-06-01 by xuankuzcr

tdcsu, 2024-03-05:
> I noticed the problem occurred with the tag of Mid360 points.
> 
> It works well for me after commenting out the following line, but I don't know what this line means.
> 
> https://github.com/hku-mars/FAST-LIVO/blob/a8918c23dc4dba7d98fa723a9fa0fc2cbdf4d977/src/preprocess.cpp#L139
> 
> [image]

Thanks for your tips, could you please tell me your sensor type of the camera