hku-mars/FAST-LIVO #57: 源码支持VELO32线雷达吗？

bzwwww, 2023-03-22:
尝试了很多自动驾驶上的数据集，VELO32线雷达都无法正常运行，请问这是数据集中数据未时间同步的原因还是源码不支持VELO32线雷达？

xuankuzcr, 2023-03-28:
关于VELO32，我不确定源码是否支持，因为并未做过相关测试，更多适配可以参考fast-lio2，在preprocess.cpp里。 
我觉得你需要check一下 1. LiDAR-IMU外参 2. 数据是否同步

bzwwww, 2023-03-30:
> 关于VELO32，我不确定源码是否支持，因为并未做过相关测试，更多适配可以参考fast-lio2，在preprocess.cpp里。 另外我觉得你需要check一下 1. LiDAR-IMU外参 2. 数据是否同步

关于数据同步，FAST-LIVO要求相机与雷达帧严格同步吗？在测试自己所录制的数据集时，相机和雷达帧并未完全同步，但保证了两帧雷达帧中间一定只有一帧相机帧，也能正常运行并取得了较好的定位效果！但在很多公开数据集中，两帧雷达帧之间有1-2帧相机帧，是否是导致进程结束的原因？

bzwwww, 2023-03-31:
关于是否支持VELO32，我对比了FAST-LIVO和fast-lio2的源码。我发现了以下问题：
在FAST-LIVO的preprocess.cpp中的270行附近，有关的数据结构都被定义为16大小。
而在fast-lio2中preprocess.cpp中的290行附近，类似的数据结构被定义为N_SCANS，我认为这应该是不能支持VELO32的原因。
我在FAST-LIVO尝试了fast-lio2中的UrbanLoco数据集，依然不能正常运行。
基于代码的整体性，希望可以得到一些将代码改为支持VELO32的建议。

xuankuzcr, 2023-06-02:
把N_SCANS改成32试试？

closed 2023-06-10 by xuankuzcr