hku-mars/FAST-LIVO #58: 关于src/preprocess.cpp中源码的疑惑与config参数的疑惑

ZRS0326, 2023-03-22:
🕛在src/preprocess.cpp 90～98行代码中的if语句，请问这段代码的作用是什么？
    **if((abs(msg->points[i].x - msg->points[i-1].x) < 1e-8) 
            || (abs(msg->points[i].y - msg->points[i-1].y) < 1e-8)
            || (abs(msg->points[i].z - msg->points[i-1].z) < 1e-8)
            || (msg->points[i].x * msg->points[i].x + msg->points[i].y * msg->points[i].y < blind)
            || (msg->points[i].line > N_SCANS)
            || ((msg->points[i].tag & 0x30) != RETURN0AND1))
        {
            continue;
        }**
    🕛在config中关于相机的以下参数作用是什么？
**cam_d0:** -0.0944205499243979
**cam_d1:** 0.0946727677776504
**cam_d2:** -0.00807970960613932
**cam_d3:** 8.07461209775283e-05

xuankuzcr, 2023-03-29:
1. 这些条件分别是：
与上一个点在x、y、z坐标轴上的差值小于1e-8 
点的x和y坐标的平方和小于"blind"（一个预设的阈值）去除较近点
点的线数（line）大于N_SCANS（一个预设的常量）
点的tag属性的第5位和第6位不是RETURN0AND1（一个预设的常量）只要0和1回波的点
2. 代表针孔相机的畸变参数

Camilochiang, 2023-04-13:
Actually this conditions produce some problems:
- When using a mid70 (that have a special minimum range of 0.05 m) and passing close to an object , all points get rejected by the condition ```((msg->points[i].tag` & 0x30) != RETURN0AND1)```
- When using different lidars the `blind` parameter have different meanings what is confusing and lack standarization.  
 
I change the above condition to this one and it work significally better
```
 double range = pl_orig.points[i].x * pl_orig.points[i].x + pl_orig.points[i].y * pl_orig.points[i].y + pl_orig.points[i].z * pl_orig.points[i].z;
 if (range < blind) || (msg->points[i].line > N_SCANS) {
    continue;
}

closed 2023-08-20 by xuankuzcr