hku-mars/FAST-LIVO #62: IMU数据的单位问题

Cc19245, 2023-04-12:
Hi, @xuankuzcr, 感谢你出色的开源工作！

我发现 NTU_VIRAL_DATASET 的 IMU 数据是正常的 m^2/s 的单位， 而 FAST-LIVO 的 private dataset 的 IMU 数据是以 9.8m^2/s 为单位进行了归一化。但是在 [IMU积分](https://github.com/hku-mars/FAST-LIVO/blob/main/src/IMU_Processing.cpp#L369) 的时候，为什么是使用初始化的 `mean_acc.norm() ` 作为模长？

我猜测是为了统一处理这两种 dataset 的不同单位的 IMU 数据？但是 `mean_acc` 是初始的 IMU 数据，在静止初始化的情况下，它的模长不是1或者9.8，主要是因为内部包含了 IMU 零偏；在运动初始化的情况下，它应该是没有意义的。因此在后面如果以 `mean_acc.norm() ` 作为模长将 IMU 数据统一恢复到 m^2/s 的单位，是否对真实的 IMU 数据进行了一些不太准确的缩放？

不知道上述我的理解是否有问题，期待你的解答，谢谢！

xuankuzcr, 2023-05-27:
`acc_avr = acc_avr * G_m_s2 / mean_acc.norm() - state_inout.bias_a;`
首先livo需要静止初始化。另外是以`G_m_s2 / mean_acc.norm()`作为模长将 IMU数据统一恢复到m/s^2单位的，这里确实是不准确的缩放，但acc_avr才是最终使用来propa的，这个误差会在update的时候被分担在bias_a里。

closed 2023-06-02 by xuankuzcr

zzwu29, 2023-12-28:
在这里看到了和我相同的困惑，mark一下~

chenxiaocongAI, 2024-02-18:
这个误差会在update的时候被分担在bias_a里,是把bias_a当成优化变量才有这个功能把。如果在优化的时候只优化pose，不考虑imu-lidar之间的外参以及bias_g，bias_a，这里对于acc_avr的计算是否需要重新考虑