hku-mars/FAST-LIVO #64: I also have the same question, but in my case I'm using a Jetson TX2

BUG423, 2023-04-24:
I also have the same question, but in my case I'm using a Jetson Orin and when i use the command
$ roslaunch fast_livo mapping_avia.launch 
the launch shows me this 

[image]

and the rviz doesn't show me the hku2.bag file.
I made the same test in my computer and it works fine, do you know how can i solve this problem?

_Originally posted by @sebastiantorres1 in https://github.com/hku-mars/FAST-LIVO/issues/33#issuecomment-1412823839_

xuankuzcr, 2023-05-22:
Hi, suggest checking the versions of eigen and opencv.

closed 2023-06-10 by xuankuzcr

BUG423, 2023-06-10:
您好，我已经收到您的来信，非常感谢！

gloryhry, 2023-06-15:
@BUG423 I also run the code on Nvidia Jetson Xavier AGX. I had the same problem as [issue](https://github.com/hku-mars/FAST-LIVO/issues/33) at first.
At last, I fixed the problem by recompile the [vision_opencv](https://github.com/ros-perception/vision_opencv) in FAST_LIVO workspace.
```bash
cd ~/catkin_ws/src
https://github.com/ros-perception/vision_opencv.git
git checkout $ROS_DISTRO
```
- Enviorment:
  - Ubuntu 18.04
  - ROS Kinetic
  - OpenCV 4.5.3 with CUDA

@xuankuzcr Another little problem, I fond it had to run two or more times `catkin_make` to compile the ros messages without errors.