hku-mars/FAST-LIVO #71: 为什么vio和lio的update不写在一个esikf里

iwander-all, 2023-08-20:
作者您好，再问一个问题哈，
现在我们已知image时间戳和lidar帧头是硬同步的，这样也可以理解为当前帧image时间戳和上一帧lidar帧尾是硬同步的，
既然数据是同步的，那么为什么不把lio和vio的esikf合并一起操作，而是拆开为2个独立的过程呢

xuankuzcr, 2023-08-20:
好问题，对于这种不同源传感器，imu propagate一次，视觉测量和雷达测量一起更新一次，还是先更新雷达，再更新视觉，一共更新两次。1. 在雷达和相机的测量噪音不相关的条件下，数学上这两种更新方式是等价的，都是紧耦合的（视觉update的时候用的是雷达update的先验分布，而不是仅仅作为pose初值），而且分成两次矩阵运算更快，这个在惯性导航领域里面有人证明过。2. 其次就是vio这边其实是个coarse-to-fine的过程的，从第2层算到第0层，要更新3次（每次都是ieskf，迭代到收敛），如果每层都把LiDAR测量拿过来一起更新的话，计算量太大了。3. 从工程设计角度来说，不存在完全对齐的相机和雷达数据，比如要是有10ms的曝光时间，你就可以让imu再propa个10ms再更新vio，但写到一起就不好处理了。分开写更方便潜在时差测试。

iwander-all, 2023-08-21:
感谢回复，醍醐灌顶，豁然开朗。

closed 2023-08-21 by iwander-all

zhh2005757, 2023-08-29:
> 好问题，对于这种不同源传感器，imu propagate一次，视觉测量和雷达测量一起更新一次，还是先更新雷达，再更新视觉，一共更新两次。1. 在雷达和相机的测量噪音不相关的条件下，数学上这两种更新方式是等价的，都是紧耦合的（视觉update的时候用的是雷达update的先验分布，而不是仅仅作为pose初值），而且分成两次矩阵运算更快，这个在惯性导航领域里面有人证明过。2. [...]

组合导航里的序贯滤波