hku-mars/FAST-LIVO #74: 关于代码中#ifdef USE_IKFOM

SlamCabbage, 2023-09-03:
大佬您好，代码中如果使用IKFOM，更新的状态是state_point = kf.get_x();
我在阅读代码的时候看到vio中的状态采用的是StatesGroup  state;
在代码laserMapping.cpp开始处：
#ifdef USE_IKFOM
esekfom::esekf<state_ikfom, 12, input_ikfom> kf;
state_ikfom state_point;
vect3 pos_lid;
#else
StatesGroup  state;
#endif
是不是意味着，如果使用IKFOM那么代码将只运行LIO呢？

hr2894235132, 2023-12-28:
> 大佬您好，代码中如果使用IKFOM，更新的状态是state_point = kf.get_x(); 我在阅读代码的时候看到vio中的状态采用的是StatesGroup state; 在代码laserMapping.cpp开始处： #ifdef USE_IKFOM esekfom::esekf<state_ikfom, 12, input_ikfom> kf; state_ikfom state_point; vect3 pos_lid; #else StatesGroup state; #endif 是不是意味着，如果使用IKFOM那么代码将只运行LIO呢？

我也注意到这个了，本来想使用IKFOM试试算法效果，LIO部分修改好后发现视觉好像不是很方便修改为IKFOM，请问你解决这个问题了吗？

PengKunPROO, 2024-01-13:
我也注意到了这个问题，而且在iekf迭代的时候，有个imgen开关，那里面只有一行for循环，意思是开源的代码和把VIO融入KF没有关系？实际上还是使用LIO的state做KF平滑位姿。。。VIO啥也没提供？就提供了颜色？但是如果VIO没有融入进去，和原论文说的也不一致啊。。。

closed 2024-06-20 by xuankuzcr