hku-mars/FAST-LIVO #76: 机械雷达是否可以使用呢？该如何使用呢？

molong-d, 2023-09-22:
作者您好！ 感谢您这么优秀的开源工作，深感敬佩，

最近正在了解多传感器融合的方向，注意到您的论文中提到机械雷达是可以此算法的。但是关于机械雷达（velodyne）如何在此算法中进行使用呢？

希望作者能够指教一下

感激不尽

closed 2024-06-20 by xuankuzcr

123295, 2024-06-25:
作者您好，请问robosense的32线机械雷达适配您的算法吗 ？

xuankuzcr, 2024-06-26:
数据可以发一下，我可以适配一下

---- 回复的原邮件 ----
| 发件人 | haonan ***@***.***> |
| 发送日期 | 2024年06月25日 21:57 |
| 收件人 | hku-mars/FAST-LIVO ***@***.***> |
| 抄送人 | Chunran ***@***.***>,
State ***@***.***> |
| 主题 | Re: [hku-mars/FAST-LIVO] 机械雷达是否可以使用呢？该如何使用呢？ (Issue #76) |

作者您好，请问robosense的32线机械雷达适配您的算法吗 ？

123295, 2024-06-26:
作者您好，我昨天用robosense的机械雷达录了包，请问这样的数据符合您的需求吗？如果不合适的话请告知我，我将尽力配合您。

bad boy
***@***.***

 
 | 发件人 | haonan ***@***.***&gt; |
 | 发送日期 | 2024年06月25日 21:57 |
 | 收件人 | hku-mars/FAST-LIVO ***@***.***&gt; |
 | 抄送人 | Chunran ***@***.***&gt;,
 State ***@***.***&gt; |
 
 

从QQ邮箱发来的超大附件

robosense32lidar.bag (188.59M, 2024年07月26日 15:40 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=2a30373522a40296c818cfa41533064d1b445553060007014e520602011e065653061a025155564f01010f020451035454035502332734100c525846565d470750025b5c5752464c015150350e&code=c075334b

123295, 2024-06-26:
作者您好，由于github网页端不能上传超过25Mb大小的数据，我已经将录的包发到您的邮箱，谢谢！