hku-mars/FAST-LIVO #80: 视觉残差对R的雅可比推导可能有误

chenxiaocongAI, 2023-10-21:
1.代码中，红框部分求导应该是错误的把。我计算出来的雅可比和你的不一致，能否帮我查下，是否我的求导有问题。谢谢啦
[image]
[image]

xuankuzcr, 2023-10-23:
这个雅各比我们数值求导去验证过，肯定没问题。你的小量怎么是左乘的，fast-livo里的状态增量都是右乘的。你在代码里把红框里换成你推导的试一下，跑一下看看效果是否变差，就知道是否有问题了。

chenxiaocongAI, 2023-10-24:
我想表达的是，你们对R的求导之后，后面一项t使用的是相机和IMU转换如下
[image]
但我这边推导使用的t是当前位姿相对世界坐标系的t，在世界坐标系表示。我的小量左乘法，是因为R^T,它是装置的。能否再次帮我确认下，你们对R的求导是否有问题。或者微信聊也可以18306676394

xuankuzcr, 2023-10-24:
没问题。

xuankuzcr, 2024-09-20:
我推了下你参考下
[image]

zhh2005757, 2024-10-02:
> 这个雅各比我们数值求导去验证过，肯定没问题。你的小量怎么是左乘的，fast-livo里的状态增量都是右乘的。你在代码里把红框里换成你推导的试一下，跑一下看看效果是否变差，就知道是否有问题了。

郑博，想请问下怎么用数值求导的方式去验证自己写的雅格比是否正确呀，有没有相关的参考资料？

xuankuzcr, 2024-10-11:
可以参考下面的代码示例
```python
import numpy as np

def f(x):
    # 定义你的函数 f(x)
    pass

def J_theory(x):
    # 定义你理论推导的雅可比矩阵
    pass

def numerical_jacobian(f, x, epsilon=1e-6):
    n = len(x)
    m = len(f(x))
    J_num = np.zeros((m, n))
    for j in range(n):
        x1 = x.copy()
        x1[j] += epsilon
        diff = (f(x1) - f(x)) / epsilon
        J_num[:, j] = diff
    return J_num

# 选择一个验证点 x
x = np.array([1.0, 2.0, 3.0])

# 计算理论雅可比和数值雅可比
J_theo = J_theory(x)
J_num = numerical_jacobian(f, x)

# 打印和比较两者的差异
print("Theoretical Jacobian:\n", J_theo)
print("Numerical Jacobian:\n", J_num)
print("Difference:\n", J_theo - J_num)
```
关注误差矩阵 J_theo - J_num 是否足够小。如果误差在数值精度范围内 (例如 1e-5 到 1e-6 之间)，则理论雅可比矩阵是正确的。

zhh2005757, 2024-10-15:
非常感谢！！

caihuirong, 2025-03-17:
> 我推了下你参考下 [image]

郑博，请问一下为什么你推的关于旋转的结果和代码里的不一样，代码里是Pic的斜对称，你这里是Pi的斜对称