hku-mars/FAST-LIVO #81: Compiling FAST-LIVO on 22.04

arihantlunawat, 2023-11-06:
I am having several issues compiling FAST-LIVO on 22.04. I know this is not officially supported but I am curious to know if anyone has tried to compile it and has been successful. Currently the compilation errors can be summarized as below,
- OpenCV 3 -> OpenCV 4 breaking changes, for example `CV_INTER_LINEAR` -> `cv::INTER_LINEAR`
- Sophus changing to header only library, double is no longer the default type, you have to specify a template parameter for every declaration of Sophus::SE3.
- ROS is no longer supported so had to work around it

zhongwei1122, 2024-05-09:
> I am having several issues compiling FAST-LIVO on 22.04. I know this is not officially supported but I am curious to know if anyone has tried to compile it and has been successful. Currently the [...]

Hello, may I ask if you have successfully compiled and run the FAST-LIVO algorithm in ROS2？I am trying to do this work.

closed 2024-06-20 by xuankuzcr

wtwdwr123, 2024-07-18:
> Hello, may I ask if you have successfully compiled and run the FAST-LIVO algorithm in ROS2？I am trying to do this work.

你成功了么？用ros2编译fast-livo