hku-mars/FAST-LIVO #84: 使用velodyne16建图效果不佳，不如fast_lio建图清晰

BG2CRW, 2024-01-26:
config文件中改成这样，实现对velodyne16的支持

common:
    lid_topic:  "/velodyne_points"
    imu_topic:  "/imu_data"

preprocess:
    lidar_type: 2 # Velodyne LiDAR
    scan_line: 16
    blind: 4 # blind x m disable

mapping:
    acc_cov_scale: 100
    gyr_cov_scale: 10000
    fov_degree:    359
    extrinsic_T: [ 0.0, 0.0, 0.28 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1]

camera:
    img_topic: /left_camera/image
    #xiyuan
    Rcl: [0.00162756,-0.999991,0.00390957,
         -0.0126748,-0.00392989,-0.999912,
          0.999918,0.00157786,-0.012681]
    Pcl: [0.0409257, 0.0318424, -0.0927219]

把img_enable设成0，只看激光部分效果

BG2CRW, 2024-01-26:
[image]

BG2CRW, 2024-01-26:
[image]

xuankuzcr, 2024-01-27:
你可以把数据bag，LiDAR IMU / LiDAR Camera外参，LiDAR IMU / LiDAR Camera时差，发给我我来看看问题在哪。

sumedhfb, 2024-04-01:
> 你可以把数据bag，Camera内参，LiDAR IMU / LiDAR Camera外参，LiDAR IMU / LiDAR Camera时差，发给我我来看看问题在哪。

@xuankuzcr I am trying to use the system with Velodyne VLP-16 and a ZED2 Camera. And I keep running into the "ERROR: process has died [pid 27955, exit code -11, cmd /home/nano/preston_ws/devel/lib/fast_livo/fastlivo_mapping __name:=laserMapping" 

While I followed the above changes in the config file, I am not sure if I am missing something. Please help me out

xuankuzcr, 2024-07-16:
> @xuankuzcr I am trying to use the system with Velodyne VLP-16 and a ZED2 Camera. And I keep running into the "ERROR: process has died [pid 27955, exit code -11, cmd [...]

Can you share the bag and parameters with me?