hku-mars/FAST-LIVO #85: 使用stm32对激光雷达与imu进行时间同步后，imu数据时间厝错乱？

rainlord2, 2024-02-06:
您好，
        我在使用stm32 对激光雷达与相机进行时间同步后，激光雷达点云数据与相机数据时间同步好了，但是imu时间厝是错乱的，stm32给出的pwm波形10hz，占空比50%，烦请帮忙看一下是什么问题导致的，多谢！

zhengcr2022, 2024-02-06:
imu时间戳具体是怎么错的，有图么

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年02月06日 15:43 |
| 收件人 | hku-mars/FAST-LIVO ***@***.***> |
| 抄送人 | Subscribed ***@***.***> |
| 主题 | [hku-mars/FAST-LIVO] 使用stm32对激光雷达与imu进行时间同步后，imu数据时间厝错乱？ (Issue #85) |

您好，
我在使用stm32 对激光雷达与相机进行时间同步后，激光雷达点云数据与相机数据时间同步好了，但是imu时间厝是错乱的，stm32给出的pwm波形10hz，占空比50%，烦请帮忙看一下是什么问题导致的，多谢！

rainlord2, 2024-02-18:
如下图所示，此时采用1hz pwm波进行驱动，secs的时间戳会从22 跳回到21 ，甚至有的时候会一次性跳好多秒;有没有可能是激光雷达固件版本不对，如果有可能，请问一下，你的激光雷达的固件版本是多少？
[image]

xuankuzcr, 2024-02-19:
好像和固件版本关系不大，你能给我看下你的livox_ros_driver的终端界面显示么

rainlord2, 2024-02-23:
@xuankuzcr 没明白你想看的是哪个界面，能详细说一下吗？谢谢！

xuankuzcr, 2024-05-31:
您好，请尝试下我们更新的stm32固件以及同步方案
https://github.com/sheng00125/LIV_handhold