hku-mars/FAST-LIVO #89: 如何自制手持扫描设备

MisEty, 2024-03-19:
您好，感谢您开源如此优秀的工作来推动社区发展！现在我希望能搭建类似的LIV手持扫描设备复现FAST-LIVO，但对于硬件组成仍然有一些疑问：
1. 硬件系统的具体工作流程是怎样的？我的个人理解是STM32给雷达和相机分别发送同步的脉冲信号，雷达和相机接收信号后以同步的频率进行扫描，将结果传输给计算机执行后续SLAM任务
2. 设备中的on-board PC是必要的吗？是不是所有计算都在DJI manifold-2c上执行？能否使用笔记本电脑等算力更强的设备替代？
3. 是否一定要集成ros系统？我注意到所有的驱动都是集成到ros中的，FAST-LIVO也是以ros包的形式发布的，能否类似ElasticFusion那样作为不依赖ros的视线？
4. STM32在这个系统中是必要的吗，如果只是控制脉冲同步的话能否采用更简单的板子？
5. 如果使用其他的硬件，比如其他的相机型号，能否继续采用这个设计方案？

xuankuzcr, 2024-05-31:
1. STM32给雷达和相机分别发送上升沿对其的1Hz和10Hz的方波，同时通过ttl转usb向电脑发送1Hz的人工生成的GPRMC消息，保证LiDAR和相机都同步在stm32的时钟下，并让LiDAR和相机的帧头对齐。。注：LiDAR SDK内部会将LiDAR消息10分频，并把GPRMC中的时间戳赋值给LiDAR消息。
2. 不是必要的，可以换别的
3. 可以不依赖ros
4. 可以用esp32或者arduino这种，只要时钟精度够就可以
5. 没问题，只要相机支持pps触发输出就可以

MisEty, 2024-05-31:
> 1. STM32给雷达和相机分别发送上升沿对其的1Hz和10Hz的方波，同时通过ttl转usb向电脑发送1Hz的人工生成的GPRMC消息，保证LiDAR和相机都同步在stm32的时钟下，并让LiDAR和相机的帧头对齐。。注：LiDAR SDK内部会将LiDAR消息10分频，并把GPRMC中的时间戳赋值给LiDAR消息。
> 
>     2. 不是必要的，可以换别的
> 
>     3. 可以不依赖ros
> 
>     4. 可以用esp32或者arduino这种，只要时钟精度够就可以
> 
>     5. 没问题，只要相机支持pps触发输出就可以

您好，我现在尝试不使用ros实现，已经实现了stm32的脉冲控制和授时，但不知道如何通过lidar sdk内部进行10分帧，请问livox mid360如何实现这一步呢

closed 2024-06-20 by xuankuzcr