hku-mars/FAST-LIVO #90: omp_set_num_threads(MP_PROC_NUM);             #pragma omp parallel for

chenxiaocongAI, 2024-03-19:


xuankuzcr, 2024-05-31:
只有构建约束和计算雅各比的时候是多线程的

chenxiaocongAI, 2024-05-31:
再请教一个问题： 相机和雷达只是时间同步，但两者触发的时间不一致。对于图像的处理，不会进行imu积分求解初值，图片跟踪效果就会变得很差吗？你们是否测试过雷达和相机两者不同时触发的效果呢

发自我的iPhone

chenxiaocongAI, 2024-05-31:
对于海康相机光圈很小，按着你的建议曝光0.5ms，此时相机的RGB很黑，你们是否进行了补光，补光器件是否有介绍

发自我的iPhone

xuankuzcr, 2024-05-31:
fast-livo的方法是触发同步，两者帧头时间是一致的。 对于非同时刻采集数据，我们也测试过没什么问题，视觉部分可以拉回来，可以关注下fast-livo2。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年05月31日 11:21 |
| 收件人 | hku-mars/FAST-LIVO ***@***.***> |
| 抄送人 | Chunran ***@***.***>,
Comment ***@***.***> |
| 主题 | Re: [hku-mars/FAST-LIVO] omp_set_num_threads(MP_PROC_NUM); 一旦打开，LIO部分，计算点到面会使用多个线程，这是合理的，但为什么在IEKF求解位姿的过程也是多个线程各自进行求解呢？ (Issue #90) |

对于海康相机光圈很小，按着你的建议曝光0.5ms，此时相机的RGB很黑，你们是否进行了补光，补光器件是否有介绍

发自我的iPhone

xuankuzcr, 2024-05-31:
曝光的建议是室外5ms，室内10ms内自动曝光。如果要补光的话，推荐功率较大的面阵光源。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 发送日期 | 2024年05月31日 11:21 |
| 收件人 | hku-mars/FAST-LIVO ***@***.***> |
| 抄送人 | Chunran ***@***.***>,
Comment ***@***.***> |
| 主题 | Re: [hku-mars/FAST-LIVO] omp_set_num_threads(MP_PROC_NUM); 一旦打开，LIO部分，计算点到面会使用多个线程，这是合理的，但为什么在IEKF求解位姿的过程也是多个线程各自进行求解呢？ (Issue #90) |

对于海康相机光圈很小，按着你的建议曝光0.5ms，此时相机的RGB很黑，你们是否进行了补光，补光器件是否有介绍

发自我的iPhone

chenxiaocongAI, 2024-05-31:
数据非同时触发，图片在没有imu递推作为初值情况下,它的追踪效果也能保证？这在机器人快速运动下，位姿会进入到局部最优吧。另外例如图片刚好就进入到两侦雷达数据的中间(10hz)，也是极端情况

发自我的iPhone

 
 | 发件人 | ***@***.***&gt; |
 | 发送日期 | 2024年05月31日 11:21 |
 | 抄送人 | Chunran ***@***.***&gt;,
 Comment ***@***.***&gt; |

chenxiaocongAI, 2024-05-31:
视觉部分可以拉回来，是由雷达拉回来吗？是的话，那加入的视觉岂不是起不到正作用

发自我的iPhone

xuankuzcr, 2024-05-31:
数据不同时触发也有imu prior啊......你没看懂我说的话，视觉部分可以起到正优化。

chenxiaocongAI, 2024-05-31:
好的，谢谢大佬啊

发自我的iPhone

chenxiaocongAI, 2024-05-31:
再问个问题: 记当前图片帧t1，上次雷达帧为t0，下一帧雷达帧为t2，则imu数据t0-t1用于积分作为prior的初值，然后利用图片refine。对于t2雷达的求解，则只用t1到t2之间的雷达点云数据？

发自我的iPhone

xuankuzcr, 2024-06-01:
按照现在fast-livo1的代码软同步的分帧逻辑，是用t1到t2之间的雷达点云数据的，但应该用t0到t2的雷达点云才更合理，这个fast-livo2中是这么做的。

chenxiaocongAI, 2024-06-04:
大佬有试过把AVIA替换成mid360吗？我替换后，刚体都存在1cm位资抖动，初步怀疑是vio 部分导致，还在找原因

发自我的iPhone

xuankuzcr, 2024-06-04:
#96 他的数据没有这种问题，你可以参考下。另外抖动大概率是由于同步没做好导致的。

closed 2024-06-04 by xuankuzcr

chenxiaocongAI, 2024-06-04:
> #96 他的mid360数据没有这种问题，你可以参考下。另外抖动大概率是由于同步没做好导致的。

关键相机和360静态下，也有这种跳动发生

发自我的iPhone

xuankuzcr, 2024-06-04:
IMU msg时间戳跳变 雷达帧内点时间戳跳变 雷达帧时间戳跳变 外参 你都确定没问题了么？

chenxiaocongAI, 2024-06-13:
我用你们的hku2数据集跑的时候，发现comptueJ 里面的now_error在最后开方打印出来有3点多个像素光度误差。这个误差不太正常吧，感觉有点大

发自我的iPhone

chenxiaocongAI, 2024-06-14:
请教个问题，就是你的配置文件，对于img_point_cov为什么是100或者1000。图像的协方差大于imu acc几个级别，这样VIo起到矫正imu 的累积误差是否会很有限？

发自我的iPhone

xuankuzcr, 2024-06-17:
“图像的协方差大于imu acc几个级别”，这俩noise是一个单位么？ "图像权重拉高大于imu，总体的建图效果就会下降"，如果我们不信任图像数据，那系统如何抗退化？ “你为什么在原本设置中，把img权重设置得相对较低的原因”，单位是像素误差的平方，这noise还高么？所有noise都是在满足物理意义基础上调出来的经验值，而且EKF的权重是自适应的，不存在谁高谁低这个说法，相机或LiDAR的观测比较差，对应的update就会变小。另外，我非常愿意与您交流或为您科普，但前提是保持友好的态度。

chenxiaocongAI, 2024-06-17:
sorry，我的表达可能不太友好，请原谅

chenxiaocongAI, 2024-06-17:
> sorry，我的表达可能不太友好，请原谅。一下子思路打开了.试了你退化场景bag，RGB的确起到关键作用

chenxiaocongAI, 2024-06-17:
> > sorry，我的表达可能不太友好，请原谅。一下子思路打开了.试了你退化场景bag，RGB的确起到关键作用.对于不同相机cov的设置,是否为静态像素误差的平方。还是需要通过FAST-LIVO的实际调试结果来确定这个cov

> > sorry，我的表达可能不太友好，请原谅。一下子思路打开了.试了你退化场景bag，RGB的确起到关键作用. 对于不同相机cov的设置,是否为静态像素误差的平方。还是需要通过FAST-LIVO的实际调试结果来确定这个cov

hdrive, 2025-01-02:
> 按照现在fast-livo1的代码软同步的分帧逻辑，是用t1到t2之间的雷达点云数据的，但应该用t0到t2的雷达点云才更合理，这个fast-livo2中是这么做的。

郑博你好，这里不是很明白，我看代码里边去畸变和后边 LIO 处理都是用一帧雷达点云数据的，那应该就是 t0-t2 之间所有的雷达点云数据(滤波后)，没有看到 t0-t1 之间的雷达点云数据给丢掉不用的情况，还是我的理解有误？