hku-mars/FAST-LIVO #97: Problems running your own lab dataset mid360 and ouster128

Hohaizzz, 2024-05-31:
Displayed when running ouster128 recorded dataset
[ WARN] [1717073832.500779615]: Lidar selector debug:

Displayed when running mid360 recorded dataset

[image]
[image]

I'd like to ask if no one else has had this problem, and I'd like to ask for help.

xuankuzcr, 2024-05-31:
Can you send me the bag and the camera intrinsic and extrinsic parameters for testing? I will check if there are any issues.

Hohaizzz, 2024-05-31:
mid360
[image]
ouster128
[image]

xuankuzcr, 2024-06-01:
Could you please provide me with the rosbag and the camera intrinsics?

Hohaizzz, 2024-06-03:
尊敬的作者您好，已将scr文件和所录制的bag发送至您的qq邮箱，期待您的回复

xuankuzcr, 2024-06-03:
您好，请通过邮箱zhengcr@connect.hku.hk发给我

Hohaizzz, 2024-06-03:
已发送，麻烦您了

xuankuzcr, 2024-06-03:
**This does not appear to be data from our lab. We have not publicly released any hard-synchronized data for MID360 and Ouster128.** Additionally, based on the data you provided, the camera and LiDAR are not time-synchronized. Currently, fast-livo1 only supports data that has been hard-synchronized.

Hohaizzz, 2024-06-03:
好的，多谢作者，我后面将尝试将相机和雷达时间进行同步

closed 2024-06-04 by xuankuzcr