hku-mars/FAST-LIVO #104: robosense Helios32线激光雷达适配问题

123295, 2024-06-26:
作者您好，感谢您的开源工作。请问您的算法适配robosense Helios32线激光雷达吗？
激光雷达的具体信息如下：https://www.robosense.cn/rslidar/RS-Helios，具体型号为 Helios 32 26° FoV。

xuankuzcr, 2024-06-26:
把LiDAR+imu+camera(optional)数据record到bag里，以及外参发给我，参考#37。

123295, 2024-06-26:
作者您好，我现在还没有进行硬件上的时间同步，可能还录不到符合您要求的bag，十分抱歉。我目前的进度是是把览沃的激光雷达、robosense的激光雷达、IMU、相机都安装好到无人小车上上，并且使得这些传感器能够运行，安装您的项目的指导，后续还需要标定相机和激光雷达、三个传感器进行硬件上的时间同步。
我有两个问题想向您请教：
1、用stm32进行硬件上的时间同步是否是必须的呢？如果在中控机上（我们使用的是arm架构的orin NX）都能读取到所用传感器（雷达、相机、IMU）的时间戳，是否就不需要进行硬件上的时间同步；
2、我看了您发布在github上的制作手持设备的项目，您推荐的是海康威视的工业相机，是否只能使用这款相机（或者类似的款式）呢？我们在购置设备时买的是奥比中光的astra2相机，是类似于kinect那种，我担心我们的相机从硬件层面上就不适配fast-livo算法。
劳烦您多多指教，万分感激！！！

bad boy
***@***.***

123295, 2024-06-26:
> 目前版本的fast-livo1是必须要做硬同步的，之后可能会有LIV joint initialization的功能来初始化时差。如果你能在orin nx里面读到上升沿时刻的时间戳的话，也可以称之为硬同步。硬同步不一定要LiDAR scan和image frame之间严格对其，而是二者的时间戳是在一个时间轴上，且是准确的。你可以用arm板产生脉冲信号，并读取上升沿时间。
>
> 可以用任意种类的相机，只要他具有trigger的接口。“奥比中光的astra2相机” 这个你要去手册里查一下，是否提供触发输出的接口。

十分感谢您的指导和意见，我会认真思考的！祝您一切顺利！

bad boy
***@***.***

closed 2024-06-28 by xuankuzcr