hku-mars/FAST-LIVO #105: 内存泄漏导致运行时间不能太长

JJ353, 2024-06-27:
作者您好！我现在使用的是16g内存加上8g的交换内存，运行时实际使用的内存比这稍微小一点，会遇到 #29 中的情况，制图的进程会在三分钟左右因为内存使用完而崩溃，没法运行太长时间。运行过程中内存一直在减少没有释放过，我试着关闭了img_enable，通过对比发现了因该是图像的原因，这部分会一直使用内存。如果只增加交换内存也只能稍微增加一点运行时间，还是会遇到内存泄露的问题，不能做到长时间工作。
这张图片是打开了img_enable的情况，当内存使用完就会崩溃：
[image]
这是关闭了img_enable的情况，内存一直比较稳定：
[image]
请问怎么做才能让他能够长时间运行呢？期待您的回复，万分感激！

xuankuzcr, 2024-06-27:
感谢您的关注！这个仓库的代码是我三年前开发完并在两年前完成开源的，visual module确实存在内存泄漏，一般情况是比如 list<Feature *> obs_这种数据进行删除时只删除了指针，而没有先释放指针指向的内存区域，再swap掉vector。
```cpp
  for (auto it = obs_.begin(), ite = obs_.end(); it != ite; ++it)
  {
    delete(*it);
  }
  obs_.clear();
```
但我现在更多精力在[fast-livo2](https://github.com/hku-mars/FAST-LIVO2)的代码整理上，保证代码简洁可读性强，更多种类的LiDAR和camera模型的适配，更多隐式/显式三维重建和机器人的application，以及对内存泄漏的处理上（已全部fix）一旦文章accept就开源。对于fast-livo1, 你可以优先check下我刚才说的类似问题，其他具体的内存泄漏问题，可以在launch文件中使用`launch-prefix="valgrind --leak-check=full"`去定位问题，即 `<node pkg="fast_livo" type="fastlivo_mapping" name="laserMapping" output="screen" launch-prefix="gdb -ex run --args">`

JJ353, 2024-06-27:
好的，谢谢您出色的工作和回复！

closed 2024-06-27 by xuankuzcr