hku-mars/FAST-LIVO #107: 跑kitti数据集时遇到的

zszz3, 2024-07-11:
[image]
您好，我这里使用FAST-LIVO模型跑kitti数据集，“kitti_2011_09_30_drive_0016_synced.bag”，
遇到了问题发现相机的尺寸和图片的尺寸不同，这里我通过修改相机的尺寸后发现效果非常差。请问有什么好的修改方法么，以及想问问如果我想在该模型跑kitti数据集，有什么需要修改的参数嘛（初学者不是特别懂2333）

zszz3, 2024-07-13:
顺便想问问如何保存轨迹

xuankuzcr, 2024-07-14:
很遗憾目前FAST-LIVO并没有做关于kitti数据的测试，而且kitti数据的imu有同步问题，你应该使用匀速运动模型prior，这些在FAST-LIVO中并没有实现。可以关注下即将开源的[FAST-LIVO2](https://github.com/hku-mars/FAST-LIVO2)，里面可以应用匀速运动模式，而且在kitti上做过相关测试。保存轨迹的话将state中的pos_end存成tum格式就好。可以把下面的代码段加在LIO state更新之后。
```cpp 
static bool pos_opend = false;
static int ocount = 0;
std::ofstream outFile, evoFile;
if (!pos_opend) 
{
  evoFile.open(std::string(ROOT_DIR) + "Log/result/" + hilti_seq_name + ".txt", std::ios::out);
  pos_opend = true;
  if (!evoFile.is_open()) ROS_ERROR("open fail\n");
} 
else 
{
  evoFile.open(std::string(ROOT_DIR) + "Log/result/" + hilti_seq_name + ".txt", std::ios::app);
  if (!evoFile.is_open()) ROS_ERROR("open fail\n");
}
Eigen::Matrix4d outT;
Eigen::Quaterniond q(_state.rot_end);
evoFile << std::fixed;
evoFile << LidarMeasures.last_lio_update_time << " " << _state.pos_end[0] << " " << _state.pos_end[1] << " " << _state.pos_end[2] << " "
        << q.x() << " " << q.y() << " " << q.z() << " " << q.w() << std::endl;```

closed 2024-07-15 by xuankuzcr

Jacke1ine, 2025-07-24:
> [image]您好，我这里使用FAST-LIVO模型跑kitti数据集，“kitti_2011_09_30_drive_0016_synced.bag”，遇到了问题发现相机的尺寸和图片的尺寸不同，这里我通过修改相机的尺寸后发现效果非常差。请问有什么好的修改方法么，以及想问问如果我想在该模型跑kitti数据集，有什么需要修改的参数嘛（初学者不是特别懂2333）

你好，请问你成功跑通过kitti数据集吗？