hku-mars/FAST-LIVO #110: 位姿估计只使用了可见光相机，如果使用红外相机做位姿估计的话，需要多考虑什么问题吗？

Xujiaming1, 2024-07-27:
感谢作者的优秀的开源工作！我们复现了您的fast-livo的软硬件方案，用了一个红外相机和一个可见光相机以及一个Livox Avia LiDAR，位姿估计只使用了可见光相机，但通过把红外相机和可见光相机的颜色一起加权融合来给点云上色，算法跑的效果不错如下图。我想问下如果使用红外相机做位姿估计的话，在计算光度学误差时需要多考虑什么东西么？
[image]
[image]
[image]
[image]

xuankuzcr, 2024-07-27:
感谢您的关注~感觉可以预先标定图像的亮度值和温度值之间的关系，使用温度值计算不同patch之间的光度误差。
```python
import numpy as np

# 定义普朗克常数、光速和玻尔兹曼常数
h = 6.626e-34  # 普朗克常数 (J·s)
c = 3.0e8      # 光速 (m/s)
k_B = 1.381e-23 # 玻尔兹曼常数 (J/K)

# 红外相机校准参数（示例值）
R = 1.0
B = 1.0
F = 1.0

def brightness_to_temperature(brightness, R, B, F, wavelength):
    """
    将亮度值转换为温度值
    :param brightness: 红外图像中的亮度值
    :param R: 相机校准参数 R
    :param B: 相机校准参数 B
    :param F: 相机校准参数 F
    :param wavelength: 波长 (米)
    :return: 温度值 (K)
    """
    # 普朗克定律的逆运算，求解温度
    temperature = (h * c) / (wavelength * k_B * np.log((2 * h * c**2) / (brightness * wavelength**5) + 1))
    return temperature
```

Xujiaming1, 2024-07-27:
好的，再次感谢作者优秀的开源工作和回复！

closed 2024-07-28 by Xujiaming1

85256638, 2024-11-07:
> 好的，再次感谢作者优秀的开源工作和回复！
@Xujiaming1 请问佬用的什么红外相机？把红外相机也融合起来，在代码里需要改动哪里呢？谢谢！

Xujiaming1, 2024-11-12:
你好，红外相机使用的是海康的，型号：MV-CI003-GL-N15。