hku-mars/FAST-LIVO #111: 关于单开LiDAR模式不如FASTLIO2的浅谈

Gatsby23, 2024-08-07:
今天测试发现FASTLVIO的精度略差于FASTLIO2，debug后感觉应该是如下几个方面：
1. LiDAR本身的特征提取模块，velodyne_handler不够好，建议机械激光雷达的采用[issue](https://github.com/hku-mars/FAST-LIVO/issues/99) 中提到的XT-32，机械激光雷达的特征则基本稳定。
2. FASTLIVO基于最原始的FASTLIO版本，默认LiDAR与IMU之间是单位阵，所以当外参变化的时候，效果肯定不行，大家可以参考[FASTLIO](https://github.com/chengwei0427/ESKF_LIO)加上外参估计部分。
完成以后，则效果直接起飞。

[image]

xuankuzcr, 2024-08-08:
感谢您的发现和反馈！
1. 不是特征提取的原因，而是velodyne_handler中没有考虑LiDAR点的时间戳情况。由于Hesai XT32和Velodyne点的时间戳单位都是微秒（us），因此可以混用。现在velodyne_handler已经修复。
2. 由于之前Fast-LIVO1仅在Ouster和Avia及其内置IMU下测试过，所以默认LiDAR和IMU之间的旋转是单位阵。现在已经在laserMapping.cpp中和IMU_Processing.cpp的点云去畸变部分加入了LiDAR和IMU的外参。

具体修改见这次[commit](https://github.com/hku-mars/FAST-LIVO/commit/2b27cbe).

85256638, 2024-12-05:
@Gatsby23 大佬请问你的XT-32和IMU做时间硬同步了吗？

Gatsby23, 2024-12-06:
> @Gatsby23 大佬请问你的XT-32和IMU做时间硬同步了吗？

没有，没有，理论上软同步应该效果也还可以