hku-mars/FAST-LIVO #112: LiDAR Camera no common FOV

LiuYMUNI, 2024-08-13:
Dear authors, thanks a lot for your great work. I got a question about the FOV of LiDAR and camera. Is it possible to run FAST-LIVO with a LiDAR installed at the front of the agent and camera and the tail of the agent. I mean, the two sensors do not share  FOV. Is it possible for FAST-LIVO to perform state estimation without modification in such configuration?

xuankuzcr, 2024-08-14:
Thank you for your interest. Due to the tight coupling of the camera and LiDAR at the measurement level, we do not support installation configurations where the camera and LiDAR have no overlapping FoV.

xuankuzcr, 2024-08-15:
Hi, you can close this issue if there are no further concerns.

LiuYMUNI, 2024-08-15:
> Hi, you can close this issue if there are no further concerns.

Thanks for your support and reminding.

closed 2024-08-15 by LiuYMUNI