hku-mars/FAST-LIVO #115: Problems with the accelerometer and gyroscope bias estimation

MatteoMariani99, 2024-08-21:
Good morning, and thank you for sharing your work. 
I am working with an OUSTER OS0-128 lidar (with an internal IMU) and a ZED2 camera. All three sensors are synchronized with the same timestamp and the camera is positioned above the lidar. 
[image]

I tried running the FAST-LIVO algorithm by modifying the avia_resize.yaml (zed2_resize.yaml attached file) file as you suggested. 

[image]

In the first 3 seconds, the algorithm runs, and the lidar point cloud appears, but afterward, no more points are added (Add 0 3D points). Moreover, I noticed that the bias estimation for the accelerometer and gyroscope is initially set to 0 and then turns into -nan (I don't know why). 
Could you provide an explanation for this?

xuankuzcr, 2024-08-22:
There could be several reasons for the issue. Could you please send me your bag and the config? I’ll help identify the problem.

MatteoMariani99, 2024-08-22:
Thank you for your fast reply! I have uploaded everything to the following Drive link: 
https://drive.google.com/drive/folders/1NkM0H5jK0iF_742M52Ox6PUHOjkEF7SA?usp=drive_link

xuankuzcr, 2024-08-22:
I've already requested download access. Could you please approve it?

MatteoMariani99, 2024-08-22:
I have approved it.

MatteoMariani99, 2024-08-29:
Thank you for your response. I will check in the next few days and then let you know.

MatteoMariani99, 2024-08-29:
Is it possible to use your LIDAR_IMU_init repository for the time synchronization of the lidar and the IMU?