hku-mars/FAST-LIVO #116: Timestamp Synchronization Question

yunlongwangsd, 2024-08-21:
首先感谢hku-mars实验室的各位做出如此惊艳的项目！！
我的测试平台是Jetson Orin + Livox MID 360 + Leopard stereo hawk camera. 
请问激光雷达和相机的timestamp需要完全准确对齐吗？（指rostopic echo 里的timestamp必须一模一样嘛）我现在是使用ptp 来同步激光雷达和Jetson Orin, 不知道是否一定需要用pps信号来同步相机。 
谢谢！！

xuankuzcr, 2024-08-22:
您好，感谢您的支持和关注！不需要完全对齐，只要相机和LiDAR准确的采样时刻都在一个时间轴上就可以了。

yunlongwangsd, 2024-08-22:
感谢您的回复！

请问都在一个时间轴的意思是说相机和LiDAR即使采样频率不同，但只要都同步到同一个时钟就可以？

当前遇到的同步问题是因为相机的数据是通过ROS Bridge从ROS2 传到ROS1， 过程中相机会有明显丢帧，导致`FAST-LIVO`的运行结果的delay随着时间越来越严重。请问是因为传输过程中相机丢帧导致的delay，还是因为没有将相机和LiDAR同步到同一个时钟所导致的delay问题呢？

xuankuzcr, 2024-08-22:
“请问都在一个时间轴的意思是说相机和LiDAR即使采样频率不同，但只要都同步到同一个时钟就可以？” 是的。
“请问都在一个时间轴的意思是说相机和LiDAR即使采样频率不同，但只要都同步到同一个时钟就可以？” 有可能是因为丢帧，具体要看数据去分析delay问题到底出在哪。

yunlongwangsd, 2024-08-23:
感谢您的回复！！
最终找到了问题关键，是因为我们的图片太大了，在我们缩小图片之后，delay问题明显好了许多，只是如果激光雷达的采集频率小于相机的采集频率的情况下，还是会有些延迟。谢谢！！

closed 2024-08-27 by yunlongwangsd