hku-mars/FAST-LIVO #117: Confuse about "flg_EKF_inited" flag.

OliverShaoPT, 2024-08-22:
for (iterCount = -1; iterCount < NUM_MAX_ITERATIONS && **flg_EKF_inited**; iterCount++)  {
   .....
   /*** Iterative Kalman Filter Update ***/
   if (**!flg_EKF_inited**){
      ...
   }
}

xuankuzcr, 2024-08-22:
It means that the EKF update should be performed after the initialization is successfully completed.

OliverShaoPT, 2024-08-22:
> It means that the EKF update should be performed after the initialization is successfully completed.

The inside "!flg_EKF_inited" condition will not be triggered, as the BIG for_loop's condition contain "flg_ekf_inited".

OliverShaoPT, 2024-08-22:
> It means that the EKF update should be performed after the initialization is successfully completed.

The inside if condition "!flg_EKF_inited" won't be triggered as the outside big for_loop's conditions contain "flg_EKF_inited".
I am confused about:
1, Is the initialization process in the EKF update loop useless? 
or
2, Should I delete the  flg_EKF_inited condition in the EKF iterated for_loop?

[image]
[image]

xuankuzcr, 2024-08-22:
You seem to be right. I think it's safe to delete it without any issues.

closed 2024-08-22 by OliverShaoPT