hku-mars/FAST-LIVO #120: 硬件设备

LixiaoyuNCEPU, 2024-08-29:
您好，请问DJI M300 RTK那个数据集使用的海康相机镜头是什么型号的呀

xuankuzcr, 2024-08-29:
您好，CMOS和镜头的具体型号和链接如下：
[CMOS](https://www.hikrobotics.com/en/machinevision/productdetail/?id=5762) MV-CA050-11UC
[Len](https://www.kowa-lenses.com/en/lm5jcm-2mp-industrial-lens-c-mount?srsltid=AfmBOoqbRu18PW8qnUeVbmJJdIWnVWeFplDzbyItqCSuDP-Tfi-cqsex) LM5JCM

LixiaoyuNCEPU, 2024-08-29:
好的，谢谢您

LixiaoyuNCEPU, 2024-09-20:
作者您好，我使用CMOS和avia运行算法，但是在rviz里显示的扫描形状和您的数据集扫描形状不一样，因为CMOS原先提供的图像是2592*2048，但在我的设备上无法运行，之后我通过海康相机的ros驱动把图像改成了1312*1024，运行出来扫描形状就变成了这样

[image]
请问如何如何能达到您数据集那样的效果呀

xuankuzcr, 2024-09-20:
扫描形状这样一般是由于LiDAR FoV大于相机的，所以被截取的相机FoV内的点云是这样的。但听你的描述，大概率是图像降采样后内参没有一起resize。如果还跑不了，可以把数据bag和标定参数发给我，我帮你测下。

LixiaoyuNCEPU, 2024-09-21:
您好，就是海康相机那个ros驱动里的参数改完以后，图像就被裁剪成左上角对应的那一部分了，之后我又重新进行了内外参标定，可以运行但是效果不太好，看您数据集提供的图像是1224×1024的，并且您的相机是2448×2048，镜头FoV和avia雷达FoV基本相差不多，扫描的形状也基本是个圆形，请问您对图像是怎么处理的呀，下面是我运行时设置的参数
[image]
[image]
[image]
``