hku-mars/FAST-LIVO #121: 模型使用的相机个数表示疑惑

chenqi13814529300, 2024-08-30:
我是个新手，我有个疑惑，hku1.bag数据有left_camera和right_camera，为什么你们只使用了left_camera一个（在aviz_reize.yam中发现），希望可以回复我一下。如果要使用两个camera我又需要额外什么工作，希望可以得到您的指点。

xuankuzcr, 2024-08-31:
感谢关注！因为本项目只关注于单相机-IMU-LIDAR的融合。如果要使用多个相机的话，需要同时使用所有相机测量，要解决耗时问题，控制每帧需要align的图像块的个数，可以参考下https://arxiv.org/pdf/2003.02014。

chenqi13814529300, 2024-09-02:
谢谢您的回答，最近我想用您出的slam数据跑3dgs，目前确实跑起来了，但是效果很差，我预计得到的image.txt和points3D.txt是错误的。

chenqi13814529300, 2024-09-02:
我的points3D.txt的数据就是您的rgb_scan_all.pcd数据，没有进行一些坐标系转换。我的image.txt的位置+四元数 使用的是pos_log.txt的数据（我将欧拉角转换为四元数了），同样没有进行其他的一些坐标系转换，使用的图片就是.bag读取得到的图片（1277张图像，但是pos_log是1264个记录，我很疑惑，我只是把最后13张图像删除了）构成了images.txt。在这基础上我也试过将points3D乘一个外参，以及将pose_log数据乘一个外参，好像都存在问题。希望可以得到您的回复，谢谢，我是个新手，有些问题可能比较愚蠢。

xuankuzcr, 2024-09-07:
应该是你的pose和image没对齐，差的13张应该是系统初始化时候的图像，你试试删除前面13张。另外检查下坐标系。

zhangqi-sudo, 2025-08-04:
> > 我是个新手，我有个疑惑，hku1.bag数据有left_camera和right_camera，为什么你们只使用了left_camera一个（在aviz_reize.yam中发现），希望可以回复我一下。如果要使用两个camera我又需要额外什么工作，希望可以得到您的指点。
> 
> 佬，你的两个相机都用上了吗，我也是个新手，不知道怎么结合这两个相机的数据，如果你解决了的话可以帮我一下吗

两位佬解决了吗